cyberbotics/webots #5832: Web Proto: double loading

BenjaminDeleze, 2023-02-03:
**Describe the Bug**
When loading a proto, the `.proto` file is loaded twice:
- Once in `loadProto` of  `ProtoManager.js`. Which seems logical.
- Once in `prepareProtoDependencies` of `Node.js` where it should not be needed as we already have it (however I guess it is more convenient to reload it in order to allow recursion)

ad-daniel, 2023-02-03:
I'm not sure to understand what you mean by loaded twice. `prepareProtoDependencies` goes through the PROTO tree and downloads all EXTERNPROTO references and it also takes care of generating the PROTO models from these files (the information is stored in `cProtoModels`). When the node building takes place, it's built from the model

BenjaminDeleze, 2023-02-03:
I mean that the `.proto` is download twice.
To see that open the developper tool and go the network tab.

(on webots.cloud it is even more that twice as we load the proto to get the documentation but it's another problem)

ad-daniel, 2023-02-03:
Note: to trigger the cascaded download of the PROTO dependencies `prepareProtoDependencies` is recursively used. It's fine for all sub-PROTO but this function is also used at the root level, for which the root file has been technically downloaded already (to trigger the recursion), so it should be possible to avoid downloading the root file twice