cyberbotics/webots #5835: robot_parallel_step fails when run by test_suite.py with debug build of test and webots

brettle, 2023-02-05:
**Describe the Bug**
When running the `robot_parallel_step` test via `test_suite.py` using a debug build of the test and of webots, I consistently get:
```
FAILURE with robot_parallel_step: Camera measurement after first pose update is wrong.
```
This does not occur when running the test from the webots GUI and does not occur if both webots and the test are release builds. I suspect a race condition.

**Steps to Reproduce**
1. `make -j18 debug`
2. `cd tests`
3. `make -C api/controllers/robot_parallel_step debug`
4. `./test_suite.py --group=api api/worlds/robot_parallel_step.wbt`
5. See failure

**Expected behavior**
No failure.

**System**
 - Fedora 37 Workstation
 - Graphics Card: NV167 / Mesa Intel® UHD Graphics 630 (CFL GT2)

stefaniapedrazzi, 2023-02-06:
Same as #5780.

brettle, 2023-02-08:
Yep. This is a duplicate. I should have searched first...

closed 2023-02-08 by brettle