cyberbotics/webots #5870: Allow images and point clouds to to be accessed during steps (i.e. between `robot_step_begin` and `robot_step_end`

brettle, 2023-02-10:
When using a local controller, memory-mapped files are used to pass images and point clouds from webots to the controller and the API calls return pointers to the associated memory. As a result the contents of the images and point clouds that the controller is accessing can change between `robot_step_begin` and `robot_step_end`, when the sensor updates the image. As of PR #5859, this behavior is documented.

This issue is to discuss changing the behavior to allow for the controller to access the data during a step. Options include:

1. Have the API return a pointer to a *copy* of the data. This would work but the extra copy would presumably hurt performance. I'm assuming a desire to avoid making such a copy was the reason for using memory-mapped files to begin with.
2. Double buffer the data. This would address the performance concern but would not entirely solve the problem. In particular, the pointer returned by an API call made after step n would not point to stable data during step n+2. That behavior is arguably even less intuitive than the current behavior.
3. Stick with the current behavior. If the controller needs to access the data during a step, it can always make a copy of the data before calling `robot_step_begin`, and access the copy.

Other ideas? I'm currently in favor of option 3.

omichel, 2023-02-10:
Yes, I agree with you that option 3 is the best. I have no better idea.

brettle, 2023-02-11:
Closing as won't fix. Reopen if there are other ideas.

closed 2023-02-11 by brettle