cyberbotics/webots #5874: Crash on inserting the same solid twice

BenjaminDeleze, 2023-02-10:
**Describe the Bug**
When inserting a Solid Proto with the same name as another Solid Proto that is already present, if its trigger the proto regeneration, Webots will crash.

The problem seems to come from #5779 

**Steps to Reproduce**
1. Open `Panda.wbt` on develop
2. Click on endEffectorSlot of the `Panda`
3. Add a `PandaHand` to it (one is already present by default)
4. Webots crash

**Expected behavior**
Webots should not crash and add the other PandaHand.

**Additional context**

- The crash happens in `WbSolid::resolveNameClashIfNeeded`. It seems to happen because `mSolidChildren` is corrupted in some way: I tried to query the size of `mSolidChildren` and just before the crash it goes crazy like 3612067496776
- I tried to comment some part of the code, and the crash was still occurring but not always at the same place, so I think that there is a memory leak or something along those line.
- Before #5779 I could insert the second PandaHand and I would get a warning stating that two solids have the same name. Now it seems like Webots is trying to rename the second one in `panda hand(1)`.
- In `Panda.proto`: 
```
endPoint DEF panda_link7 Solid {
  translation 0.088 0 0
  rotation 0.8614190923211578 -0.3591359294777342 0.3591358121139644 1.7194199439154274
  children [
    CadShape {
      url [
         "meshes/link7.dae"
       ]
    }
    %< if (fields.endEffectorSlot.value.length > 0) { >%
    DEF panda_link8 Transform {
      translation 0 0 0.107
      children IS endEffectorSlot
    }
    %< } >%
    ]
    name "panda_link7"
```
If I remove the templating (line beginning with `%<`), I got neiter a warning nor a crash nor a rename.

stefaniapedrazzi, 2023-02-13:
Given that the slot containing the newly added PROTO node is a template regeneration field, automatically changing the solid name triggers the top Panda PROTO regeneration and causes to access invalid nodes.
There is already a check to skip automatically updating the names in case this triggers the regeneration (instead a specific warning is printed) but the existing check doesn't correctly detect this case where it is not the "name" fields that is a template regeneration field but an ancestor field.

I'm now also check why the warning is no longer printed when removing template code.

stefaniapedrazzi, 2023-02-13:
Just for note, I also compared the behavior with R2023a rev1 and also in this version no warning and no rename occurs when removing the template code.

PR #5886 (2023-03-01): Fix issues resolving Solid name clash
Fix #5874 and #5842:
* crash occurs because fields/nodes causing PROTO regeneration are not correctly detected
* fix name clash not resolved (name is not updated) when inserting a PROTO parameter
* fix error in WbNode::fullPath overwriting the path (introduced in #5779)
closed 2023-03-01 by stefaniapedrazzi