cyberbotics/webots #5918: Exported images in real-time or fast mode are initially corrupted with Linux Mesa Intel® UHD Graphics 630

brettle, 2023-02-19:
**Describe the Bug**
Images exported with `wb_supervisor_export_image` in real-time or fast mode are initially corrupted on my Linux system when using the integrated Intel UHD Graphics 630. The problem does not occur under _any_ of the following circumstances:

1. Using the discrete NVIDIA card.
2. Booting the same system to Windows 10.
3. Stepping (instead of running in real-time or fast mode).
4. Running the test suite as part of CI on any of the current configurations (Ubuntu, Windows, or MacOS).
5. Reloading the world *after* the corruption occurs trying to export another image.
6. Setting the `INTEL_DEBUG` environment variable to `norbc` (or its replacement `noccs`). **This is a workaround for the problem.**

I found a [similar issue](https://gitlab.freedesktop.org/mesa/mesa/-/issues/6029) that was reported to Mesa, but so far the Mesa developers haven't determined whether the problem is with the app or Mesa.
 
**Steps to Reproduce**
1. In the master branch, run: `./webots --mode=pause tests/with_rendering/worlds/supervisor_export_image.wbt`
2. Click on the play button or the fast button.
3. See something similar to the screenshot below.

**Expected behavior**
The view should cycle through a solid red, solid green, and then solid blue and the test should pass.

**Screenshots**
[image]

**System**
 - Operating System: Fedora Linux 37 (Workstation Edition)
 - Graphics Card: Mesa Intel® UHD Graphics 630 (CFL GT2)

**Additional context**
I'm fine with just having this closed as "won't fix" on the theory that it's a Mesa bug, but I thought someone more familiar with OpenGL might be able to determine if it's actually a Webots bug. Also, I wanted to document the workaround in case anyone else runs into the same problem.

omichel, 2023-02-20:
Thank you for reporting this.
It is not a priority for us to investigate it as we do not recommend using Mesa with Webots.
However, it is good to have the issue open here as it provides useful information, including a workaround, in case anyone else bumps into the problem. It would be nice however if you could edit your post to add the version of Mesa you have on your system, as the problem might be resolved in the future with a new version of Mesa.

brettle, 2023-02-21:
This appears to be fixed with Mesa version 22.3.5.

brettle, 2023-02-21:
Correction, it's not fixed in 22.3.5. Not sure why it temporarily worked without the workaround, but it's back.