cyberbotics/webots #5920: simulationSetMode(0) causes error

Nelson-iitp, 2023-02-19:
**Describe the Bug**
When I call simulationSetMode(0), it always throws an error.

Traceback (most recent call last):
  File "superctrl.py", line 62, in <module>
    supervisor.simulationSetMode(0)
  File "C:\Program Files\Webots\lib\controller\python\controller\supervisor.py", line 55, in simulationSetMode
    self.simulation_mode = mode
AttributeError: can't set attribute

**Steps to Reproduce**
Call simulationSetMode(0) from a supervisor robot

**Expected behavior**
Not throw error, it was working fine with previous version of webots

**Screenshots**
[image]

**System**
 - Operating System: Windows 10

**Additional context**
Kindly suggest if any other info required.

omichel, 2023-02-20:
This was already fixed in #5603. Please download a [nightly build of Webots R2023a-rev1](https://github.com/cyberbotics/webots/releases) which includes a patch for this bug. Alternatively, you may also fix it in your local Webots installation by applying the patch corresponding to #5603.

closed 2023-02-20 by omichel

Nelson-iitp, 2023-02-20:
thanks, I just manually changed that line 105 in `lib/controller/python/controller/supervisor.py`. I hope it don't break other things.

omichel, 2023-02-20:
No, it should be safe.