cyberbotics/webots #5949: URDF test is not robust and fails quite often

stefaniapedrazzi, 2023-02-28:
The `robot_urdf` test in the test_suite is failing quite often but not systematically.

ad-daniel, 2023-02-28:
are you referring to the issue where it fails because it doesn't find the utility to do the check?

stefaniapedrazzi, 2023-02-28:
I got at least two non consecutive fails just yesterday.
For example it fails after restarting the tests when changing the label even if just before it worked correctly.
1. error message 
    ```
    FAILURE with robot_urdf_ur5e: URDF verification failed: "
    ```
    https://github.com/cyberbotics/webots/actions/runs/4284745624/jobs/7462821573
3. error message
     ```
     FAILURE with robot_urdf_prob3: URDF verification failed: robot name is: TIAGo++
     FAILURE with /home/runner/work/webots/webots/tests/api/worlds/robot_urdf.wbt: Timeout the results file has not been written quickly enough
     ```
     https://github.com/cyberbotics/webots/actions/runs/4282103328/jobs/7456578933
     
 I didn't inspect what is causing the failure.