cyberbotics/webots #5960: Test suite changes the graphical options to minimum

BenjaminDeleze, 2023-03-02:
Since #5806, running the test suite locally will change all graphical parameters to the minimum (disable gtao, shadows,...) which is pretty annoying when developing.

It is because we are now running "Run a copy of webots in the background to ensure doing so doesn't cause any tests to fail. Use WEBOTS_SAFE_MODE=true to ensure it always runs with the same (empty) project."

The WEBOTS_SAFE_MODE environment variable is responsible for the parameter changes https://github.com/cyberbotics/webots/blob/c759a4dbfac3addd4ded56a9a470c3d5171873fa/src/webots/gui/WbGuiApplication.cpp#L260

As the changes are made by Webots (and not directly by the test suite) it seems more complicated to revert them.

One solution would be to run this specific test only in the CI.

What do you think?

omichel, 2023-03-02:
Do you mean set the `WEBOTS_SAFE_MODE` only when running the test in the CI and not when running the tests locally?

stefaniapedrazzi, 2023-03-02:
I think we should test the default parameters instead of the WEBOTS_SAFE_MODE even in the CU, as it was the case before.
This will run more meaningful tests.

stefaniapedrazzi, 2023-03-02:
If the objective is to start Webots on the empty project, there are better options to achieve it rather than enabling the `WEBOTS_SAFE_MODE`.

PR #5965 (2023-03-06): Fix test_suite overwriting local Webots configuration
Fix #5960:
start directly the empty world instead of using the `WEBOTS_SAFE_MODE` to avoid overwriting the local Webots configuration.
closed 2023-03-06 by stefaniapedrazzi