cyberbotics/webots #6004: When I installed webots using source compilation, I couldn't start webots

SenHe524, 2023-03-20:
**Describe the Bug**
When I installed webots using source compilation, I couldn't start webots.

**Steps to Reproduce**
Here is a record of the terminal output when the source code is compiled
//******
root@25cec3013669:/usr/local/webots# make -j8
 * dependencies *
 PROTO files unchanged, no need to rebuild proto-list.xml
 * ode *
 * glad *
 * wren *
 * webots (core) *
 * controller library *
 C controller library (release)
 C++ controller library (release)
 Java controller library (release)
e # Java not installed or 'JAVA_HOME' not set, skipping Java API
 Matlab controller library (release)
 generating matlab
 * resources *
 *** translations ***
 *** projects ***
 make release libraries/qt_utils
 make release plugins/physics
 make release libraries/generic_robot_window
 make release controllers/generic
 make release plugins/robot_windows/generic_window
 copying to /usr/local/webots/lib/controller/libgeneric_robot_window.so
 make release plugins/robot_windows/generic
 * projects *
 !!! languages !!!
 *** controllers ***
 make release driver
 make release slave
 make_java release Driver
 make_java release Slave
e # Java not installed or 'JAVA_HOME' not set, skipping Java API
e # Java not installed or 'JAVA_HOME' not set, skipping Java API
 make release e-puck_position_supervisor

 !!! default !!!

 *** resources ***
 make release controllers/braitenberg
 make release libraries/vehicle
 make release controllers/ros
 make release c/car
e # ROS_DISTRO should not be a ROS2 distribution
 copying to /usr/local/webots/lib/controller/libcar.so
 make release c/driver
 linking libdriver.so
 copying to /usr/local/webots/lib/controller/libdriver.so
 make release cpp/driver
 linking libCppDriver.so
 copying to /usr/local/webots/lib/controller/libCppDriver.so
 make release cpp/car
 linking libCppCar.so
 copying to /usr/local/webots/lib/controller/libCppCar.so
 make release java
e # Java not installed or 'JAVA_HOME' not set, skipping Java API

 !!! objects !!!
 make release factory/conveyors/controllers
 make release create_wall/controllers
 make release computers/controllers
 make release television/controllers
 make release mirror/controllers
 make release traffic/controllers
 make release keyboard_conveyor_platform
[... 332 lines cut ...]
 make release automobile
 make release automobile_window
 make release autonomous_vehicle
 make release boomer
 make release radar_target_tracker
 make release racing_wheel
 make release VehicleDriver
 make release ros_automobile
e # ROS_DISTRO should not be a ROS2 distribution
e # Java not installed or 'JAVA_HOME' not set, skipping Java API

 !!! humans !!!

 *** libraries ***
 make release skin_animated_humans/libraries/bvh_util

 *** controllers ***
 make release skin_animated_humans/controllers/bvh_animation
 *******////

**System**
 - Operating System: ubuntu 20.04 in docker environment，ros2 version is foxy.
 
 This is the error I got when I tried to run webots using the linux installation guide
[image]

SenHe524, 2023-03-20:
Can someone help me point out my problem? What can I do to make webots work?

closed 2023-03-21 by SenHe524

omichel, 2023-03-21:
Did you managed to fix the problem?

SenHe524, 2023-03-21:
> Did you manage to fix the problem?

Yes, that was solved yesterday. The reason for the above error is that I downloaded the webots source code under windows environment, so the wrong file format (CR LF under windows, LF under linux) caused the executable file inside the source code to perform strange errors