cyberbotics/webots #6023: Do webots and vnc Remote desktop conflict?

SenHe524, 2023-03-28:
**Describe the Bug**
Recently I tried using vnc remote desktop access to a docker image on the server, and compiled and used webots from source in the image. With a little effort, webots worked fine, but after a day or two of using linux images, the desktop would disappear, That is stored in the associated with desktop ". ~ / local/share/applictaions "and"/usr/share/applications "in the" desktop "file will be deleted. I would like to ask whether this is related to webots and whether it is caused by conflicts between webots and vnc remote desktop.

**System**
 - Operating System: Linux Ubuntu 20.04

omichel, 2023-03-28:
I am not aware of any conflict between Webots and VNC remote desktop. Webots normally doesn't delete the folders you mentioned.

SenHe524, 2023-03-30:
> I am not aware of any conflict between Webots and VNC remote desktop. Webots normally doesn't delete the folders you mentioned.

I initially suspected that there might be a webots and VNC conflict, but for now I've basically ruled out webots causing VNC crashes. Thank you very much for your reply!