cyberbotics/webots #6032: Loading remote cadShape

BenjaminDeleze, 2023-03-30:
**Describe the Bug**
Strange bug where it works in a lot of setup but not in all:

***Work***

```
#VRML_SIM R2023b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "https://raw.githubusercontent.com/omichel/Shelf/main/protos/Shelf.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.11384659601330234 -0.3187749829101217 0.9409683644240309 5.557715208500585
  position -5.776529177279546 4.724710358949437 2.980453565305262
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Shelf {
  rotation 0 0 1 1.5708003061004252
}
```

```
#VRML_SIM R2023b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "https://raw.githubusercontent.com/omichel/Shelf/main/protos/Shelf.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.11384659601330234 -0.3187749829101217 0.9409683644240309 5.557715208500585
  position -5.776529177279546 4.724710358949437 2.980453565305262
}
TexturedBackground {
}
TexturedBackgroundLight {
}

```

and manually add the `Shelf` after

```
#VRML_SIM R2023b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "../protos/Shelf.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.11384659601330234 -0.3187749829101217 0.9409683644240309 5.557715208500585
  position -5.776529177279546 4.724710358949437 2.980453565305262
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Shelf {
  rotation 0 0 1 1.5708003061004252
}
```

***Does not work***
```
#VRML_SIM R2023b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "../protos/Shelf.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.11384659601330234 -0.3187749829101217 0.9409683644240309 5.557715208500585
  position -5.776529177279546 4.724710358949437 2.980453565305262
}
TexturedBackground {
}
TexturedBackgroundLight {
}

```

or

```
#VRML_SIM R2023b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.11384659601330234 -0.3187749829101217 0.9409683644240309 5.557715208500585
  position -5.776529177279546 4.724710358949437 2.980453565305262
}
TexturedBackground {
}
TexturedBackgroundLight {
}
```

and in both cases, add manually the Shelf after.
It result in a infinite `Download assets` popup and the `Shelf` never appear

Tested on ubuntu 20, master branch

omichel, 2023-03-30:
For me, it works in all the cases (on Windows). I have this world file:
```javascript
#VRML_SIM R2023b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.11384659601330234 -0.3187749829101217 0.9409683644240309 5.557715208500585
  position -5.776529177279546 4.724710358949437 2.980453565305262
}
TexturedBackground {
}
TexturedBackgroundLight {
}
```
And I copied the [Shelf.proto](https://github.com/omichel/Shelf/blob/main/protos/Shelf.proto) file inside the local project protos folder so that I can add it to the world file from the GUI. That works. I can also save it and then reload it and it works. After saving the file looks like this:
```javascript
#VRML_SIM R2023b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "../protos/Shelf.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.11384659601330234 -0.3187749829101217 0.9409683644240309 5.557715208500585
  position -5.776529177279546 4.724710358949437 2.980453565305262
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Shelf {
}
```
I cannot make it fail...

BenjaminDeleze, 2023-03-30:
I tested on Windows too, but I could not reproduce the bug either.
But on Ubuntu 20 it is 100% reproducible on my setup

ygoumaz, 2023-03-30:
I am currently on Windows (master branch) and could reproduce the bug:

1. Load the last world file given by @BenjaminDeleze (no Shelf in the file at all).
2. Add the Shelf proto file to the protos folder of the project.
3. In the Webots window, add the Shelf.
4. Download progress bar is stuck at 0%.

ad-daniel, 2023-03-30:
If it manages to download it once, after that point the asset is cached so the issue might not reappear unless you clear it. I confirm the issue is present in ubuntu 20

BenjaminDeleze, 2023-03-30:
> Yes, true. But this means that, despite the error message telling that the files could not be downloaded, they actually are. 

I am not sure of this because if I do:

- launch webots with the clear-cache option
- try to add a shelf
- infinite loading
- cancel loading
- close webots without saving
- open webots (without clear-cache)
- try to add a shelf again

It still does not work whereas it should if the shelf was in the cache

BenjaminDeleze, 2023-03-30:
argh, I followed the wrong track the whole time. It seems that Webots can download without any issue the CadShape...but not the mesh of the boundingObject

PR #6034 (2023-04-03): Fix mesh double loading
Fix #6032 

The meshes where sometimes loaded twice and the first downloader was deleted before it was finished, resulting in a infinite loading.

After further testing, it turns out that the issue came from `mNetworkReply` of [WbDownloader.cpp](https://github.com/cyberbotics/webots/blob/fix-mesh-double-loading/src/webots/core/WbDownloader.cpp).
When a download is triggered before the previous one is finished (because of double loading in our case), we need to delete the downloader and create a new one.
The problem is that we use `deleteLater()` for the `mNetworkReply`  [... cut]
closed 2023-04-03 by BenjaminDeleze