cyberbotics/webots #6046: Improve TIAGo model

stefaniapedrazzi, 2023-04-03:
Update the TIAGo model to the latest PAL Robotics version:
1. add a custom RGB-D camera (specifications https://shop.orbbec3d.com/Astra-S)
2. update base to latest URDF
3. change name of models from iron/steel/etc. to "TIAGo lite" (no arms), "TIAGo" (1 arm + gripper) and TIAGo ++ (2 arms + gripper)

thomaspeyrucain, 2023-09-22:
Hello Webots team,

The robotiq models are missing the coupling link that PAL added to interface the robotiq gripper with the arm of the robot

[image]

**Expected behavior**
The gripper should look like that for all robotiq gripper equipped on TIAGo:
[image]

You can find the model for the coupling link here:
- [With eletronics](https://github.com/pal-robotics/pal_robotiq_gripper/blob/master/pal_robotiq_85_gripper/pal_robotiq_85_description/meshes/visual/pal_robotiq_coupling.stl)
- [Without electronics](https://github.com/pal-robotics/pal_robotiq_gripper/blob/master/pal_robotiq_140_gripper/pal_robotiq_140_description/meshes/visual/pal_robotiq_coupling.stl)

And in [this urdf](https://github.com/pal-robotics/pal_robotiq_gripper/blob/master/pal_robotiq_85_gripper/pal_robotiq_85_description/urdf/robotiq_85_gripper.urdf.xacro#L293)

The coupling link stl model is the same for the 3 robotiq grippers

**Affected Packages**
List of affected packages:
  - [projects/devices/robotiq/protos/Robotiq2f85Gripper.proto](https://github.com/cyberbotics/webots/blob/8aba6eaae76989facf3442305c8089d3cc366bcf/projects/devices/robotiq/protos/Robotiq2f85Gripper.proto#L96)
  - [projects/devices/robotiq/protos/RobotiqEPickGripper.proto](https://github.com/cyberbotics/webots/blob/8aba6eaae76989facf3442305c8089d3cc366bcf/projects/devices/robotiq/protos/RobotiqEPickGripper.proto)
  - [projects/devices/robotiq/protos/Robotiq2f140Gripper.proto](https://github.com/cyberbotics/webots/blob/8aba6eaae76989facf3442305c8089d3cc366bcf/projects/devices/robotiq/protos/Robotiq2f140Gripper.proto)

**System**
 - Webots Version: R2023b
 - ROS Version: Humble
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: Nvidia

thomaspeyrucain, 2023-09-22:
We also updated the stl of the omni_base to remove the unnecessary inside components [here](https://github.com/pal-robotics/omni_base_robot/blob/melodic-devel/omni_base_description/meshes/base/base_link.stl):
[image]

BenjaminDeleze, 2023-10-03:
Dear @thomaspeyrucain,

I added the couple link to the robotiq grippers, you can check it in [this PR](https://github.com/cyberbotics/webots/pull/6401)

However, I am affraid we will not be able to use the new mesh you provided for the omni_base as it does not meet the Webots quality standard:
  - The mesh is still to big: in Webots no mesh should exceed 21845 triangles. Dividing a large mesh into several smaller ones is a possible solution, but simplifying each mesh should be done whenever possible.
  - Also, the mesh contains some artifacts like on the picture below.

[image]

thomaspeyrucain, 2023-10-03:
Hello @BenjaminDeleze ,
Thank you I will see internally what we can do about this

closed 2023-10-05 by BenjaminDeleze