cyberbotics/webots #6049: Webots integrated with matlab is not working. Nothing happens and eventually closes.

Kiesmith, 2023-04-03:
**What I am doing:** 

Making my own controller for a robot in matlab code format. 

**Problem:**
Matlab.exe opens but after a while closes. 

What I have done: 
matlab.exe is in tools - preferences - MATLAB command
minGW64 installed and set to main in matlab and verified as main in matlab command window on original software. 
When I click on matlab.exe, it opens matlab as normal. 
Matlab version: R2021a
webot version: R2023a
Matlab version is 64 bit and webots version is 64 bit. 
When running webots and running the simulation, I get this in the command window: 
INFO: epuck_go_forward: Starting controller: "D:\Matlablocation\bin\matlab.exe" -nosplash -nodesktop -minimize -sd D:/webots/Webots/lib/controller/matlab -r launcher INFO: 'epuck_go_forward' controller exited successfully.

**Screenshots to make it clear**
[image]
**Matlab.exe opens but eventually closes. nothing happens.** 

[image]
Matlab.exe put in file path. (I had not 'MATLAB.exe', only 'matlab.exe').

[image]
**mex compiler setup**

If this issue can be resolved, I believe this would greatly benefit the documentation here: https://cyberbotics.com/doc/guide/using-matlab

Thankyou, and best regards, 
Kieran.

ygoumaz, 2023-04-04:
If the controller is exiting successfully, it generally means that the controller was run and that the end of the code was reached. Do you have any step function in your MATLAB controller? Could you please provide your controller code, so that I can further investigate this issue?

Kiesmith, 2023-04-04:
What I can show you is this
[image]
The code is
`% time in [ms] of a simulation step
TIME_STEP = 64;

MAX_SPEED = 6.28;

% initialize devices
ps = [];
ps_names = [ "ps0", "ps1", "ps2", "ps3", "ps4", "ps5", "ps6", "ps7" ];

for i = 1:8
  ps(i) = wb_robot_get_device(convertStringsToChars(ps_names(i)));
  wb_distance_sensor_enable(ps(i), TIME_STEP);
end

left_motor = wb_robot_get_device('left wheel motor');
right_motor = wb_robot_get_device('right wheel motor');
wb_motor_set_position(left_motor, inf);
wb_motor_set_position(right_motor, inf);
wb_motor_set_velocity(left_motor, 0.0);
wb_motor_set_velocity(right_motor, 0.0);

% feedback loop: step simulation until receiving an exit event
while wb_robot_step(TIME_STEP) ~= -1
  % read sensors outputs
  ps_values = [];
  for i = 1:8
    ps_values(i) = wb_distance_sensor_get_value(ps(i));
  end

  % detect obstacles
  right_obstacle = ps_values(1) > 80.0 | ps_values(2) > 80.0 | ps_values(3) > 80.0;
  left_obstacle = ps_values(6) > 80.0 | ps_values(7) > 80.0 | ps_values(8) > 80.0;

  % initialize motor speeds at 50% of MAX_SPEED.
  left_speed  = 0.5 * MAX_SPEED;
  right_speed = 0.5 * MAX_SPEED;
  % modify speeds according to obstacles
  if left_obstacle
    % turn right
    left_speed   = 0.5 * MAX_SPEED;
    right_speed  = -0.5 * MAX_SPEED;
  elseif right_obstacle
    % turn left
    left_speed  = -0.5 * MAX_SPEED;
    right_speed = 0.5 * MAX_SPEED;
  end
  % write actuators inputs
  wb_motor_set_velocity(left_motor, left_speed);
  wb_motor_set_velocity(right_motor, right_speed);

  % if your code plots some graphics, it needs to flushed like this:
  drawnow;
end`

This code can be sourced from: https://cyberbotics.com/doc/guide/tutorial-4-more-about-controllers?tab-language=matlab

From the e-puck_matlab
[image]
[image]
The matlab command window closes after.

The matlab.exe command path is also in a different folder compared to the original matlab path. I cannot also find the default matlab path as stated in the link at: Windows: C:\Program Files\MATLAB\R20XXx\bin\win64\MATLAB.exe (MATLAB does not exist here and does not seem findable).

Kiesmith, 2023-04-04:
**Update - Solved**

**Cause**
The cause of the problem was using the matlab path ".../bin/matlab.exe"

**Solution**
The solution was to use MATLAB.exe at the matlab path ".../bin/win64/MATLAB.exe"

Whenever I searched for the MATLAB.exe in the files, I was only shown the one from ".../bin" and not the one from ".../bin/win64" had never appeared in the search results. This led to my problem. ".../bin/win64/MATLAB.exe" also appeared in a very large directory which was also hard to find. I found this by searching "matlab" in the file search bar and then right-clicked it to access the location within that directory. Following that I copied the path into the webot - tools - preferences. 

Still unsure why the "/bin/matlab.exe" did not work  as this was a 64 version too. Would be curious to know why and I hope this solution contributes to any issues people may face. 

[image]

Now that I solved this, I am a little unsure why it says libraries are not found (unless this refers to matlab not being open in the beginning)? 
Thankyou for your help.

ygoumaz, 2023-04-04:
Glad to hear that you were able to solve your issue. As stated in the documentation link that you sent (https://cyberbotics.com/doc/guide/using-matlab):

> On Windows, there are two MATLAB.exe files: one is located in "bin\MATLAB.exe" and the other one in "bin\win64\MATLAB.exe". "bin\win64\MATLAB.exe" should be used in the preferences, because "bin\MATLAB.exe" is just a launcher that causes problems with stdout/stderr streams and with the termination of the process.

So this is indeed the correct way to fix your issue. 

I can't reproduce the library issue on my side. Is it present at the given path? If you installed Webots from sources, is the library compiled?

closed 2023-04-05 by ygoumaz