cyberbotics/webots #6056: software crashes

warmsh, 2023-04-07:
**Describe the Bug**
When I click Save Current World File, the software crashes. The world files used are considered the official pioneer3at examples.The software version is 2023a。

**System**
 - Operating System: [Linux Ubuntu 20.04.6]
 - Graphics Card: [ AMD Radeon 680M]

stefaniapedrazzi, 2023-04-11:
Could you please try if you get the same error using the Webots R2023a revision 1 from the nightly builds?
https://github.com/cyberbotics/webots/releases/tag/nightly_10_4_2023

We already fixed recently some issues with the save world functionality.
If this should not fix your issue, then you should provide additional information because the problem is very likely specific to your system:
* which Webots package are you running? Debian, tarball package, snap?
* where is Webots installed (for Debian and tarball)? 
* where do you try to save the world?

stefaniapedrazzi, 2023-06-12:
Closed because of inactivity.

closed 2023-06-12 by stefaniapedrazzi