cyberbotics/webots #6058: Crash when wb_robot_wtep caused by setMFVec3f for point Field of IndexedLineSet Node

dooxu9527, 2023-04-07:
I'm trying to draw the swing trajectories of quad robot's four feet, like what supervisor_draw_trail does.

What I do is as below:
1. Create Shape Node by string like this in world
[image]
2. Get position from my controller & draw it by pointField->setMFVec3f([positions]) and coordField->setMFInt32([coords])

My problem:
It goes well when I just draw one line, but if I draw more lines it works well in the beginning and will **crash after some time**,  by which the software closes directly. 
If I draw **two lines** it will **crash after about 1min** in simulation, and it will **crash after about 30s** if I draw **four lines**. 

My test：
1. If I only comment out the call of **setMFVec3f**, it won't crash, so I guess there are something wrong with it.
2. When I run the simulation using gdb, it shows that the iteration that the code stoped at wb_robot_step(TIME_STEP), see the pictures:
[image]
[image]

stefaniapedrazzi, 2023-04-11:
Could you please submit your world and controller file?
I tried to reproduce it starting from the `supervisor_draw_trail` example using a C and C++ controller but everything works fine for me.

Which version of Webots are you using?
Could you please try the latest R2023 revision 1 available from the nightly builds?
https://github.com/cyberbotics/webots/releases/tag/nightly_10_4_2023

dooxu9527, 2023-04-12:
> > Could you please submit your world and controller file? I tried to reproduce it starting from the `supervisor_draw_trail` example using a C and C++ controller but everything works fine for me.
> > Which version of Webots are you using? Could you please try the latest R2023 revision 1 available from the nightly builds? https://github.com/cyberbotics/webots/releases/tag/nightly_10_4_2023
> 
> Thank you for reply! I have tried the R2023a-rev1 and it works well! The version I used was R2023a_amd64, and I found that my system is x86 not amd64, so this may be the reason.

I have tried the stable version webots-R2023a-x86-64 back, but it doesn't work again. So only the pre-release R2023a-rev1 version doesn't have this problem.

stefaniapedrazzi, 2023-04-17:
Thank you for the feedback!
We already applied many bug fixed to R2023a and included in R2023 version1.
So it seems that also this bug has already been fixed.

closed 2023-04-17 by stefaniapedrazzi