cyberbotics/webots #6059: Mecanum wheels only going forward or backward.

BMT-Jonas, 2023-04-07:
I have a mecanum robot (the two that are standard in webots: kuka and fabtino)
they both behave the same, when i run all my wheels in the same direction the robot goes forwards or backwards, all good.
But when i run my left wheels backwards and my right wheels forward which should turn my robot it starts turning, then it stops, then starts again and so on.

See my issue: https://vimeo.com/815673695

is this a bug with friction ?

BMT-Jonas, 2023-04-08:
contactProperties [
    ContactProperties {
      material1 "InteriorWheelMat"
      coulombFriction [
        0, 2, 0
      ]
      frictionRotation -0.785398 0
      bounce 0
      forceDependentSlip [
        10, 0
      ]
    }
    ContactProperties {
      material1 "ExteriorWheelMat"
      coulombFriction [
        0, 2, 0
      ]
      frictionRotation 0.785398 0
      bounce 0
      forceDependentSlip [
        10, 0
      ]
    }
  ]

solved it

closed 2023-04-08 by BMT-Jonas

cutterpocket-cloud, 2026-01-01:
Hi Jonas,
I just have the same problem like you, actually Webots R2025a is a bit different as the version before.
Do you have maybe a idea for me? Thanks a lot!
Best Regards
Peter