cyberbotics/webots #6060: Template importing error when adding new objects

CurryYH, 2023-04-08:
## Template importing error when adding new objects (After downloading Webots, open the new world for the first time and it will report this error)

**Steps to Reproduce**
1. Click on 'adding new object or import a new object'
2. Scroll down to any new objects
3. Click 'import'
4. See error

**Expected behavior**
The new object should appear in the world。

**Screenshots**
[image]

[image]

**System**
 - Operating System: [ Windows 11]
 - Graphics Card: [NVIDIA GeForce RTX 3060 6 GB]

**Additional context**
The user is currently out of China, so there is no problem accessing the Github content theoretically.

CurryYH, 2023-04-08:
Encountered the same problem in Webots  r2022a , r2022b and r2023a

stefaniapedrazzi, 2023-04-11:
This issue is very specific to the user system.
It was already reported some months ago by another user on Windows, but we were not able to reproduce it and the user didn't provide enough information to debug it.
Unfortunately without additional information we cannot help very much.

We could generate a debug Webots package and ask you to test and report the output to try to better understand what is the issue.

CurryYH, 2023-04-11:
OK, I can test the debug package and report the output to help you to debug it. By the way, what kind of additional information do you need?

CurryYH, 2023-04-11:
> 
OK, I can test the debug package and report the output to help you to debug it. By the way, what kind of additional information do you need?

stefaniapedrazzi, 2023-04-24:
I just generated a test package for Windows.
You can find it in this page at the bottom of the page, the package is named "build-windows-2019 ".
Do you see it?

You should install it, start Webots from the CMD terminal and report the output that is written by Webots in the terminal.

CurryYH, 2023-05-09:
> I just generated a test package for Windows. You can find it in this page at the bottom of the page, the package is named "build-windows-2019 ". [...]

Sorry, I have just missed the GitHub message, and the package "Build-windows-2019" has now expired, could you please reupload the package?

stefaniapedrazzi, 2023-05-09:
I just regenerated the package.

stefaniapedrazzi, 2023-05-10:
That's because you would need to call `webots-bin.exe` instead of `webotsw.exe`.
However, given that it can be tricky to setup correctly the environment manually to execute the binary directly, I changed the debug package so that it prints the data in the Webots terminal.

I would ask you to download and install the latest package from this link:
https://github.com/cyberbotics/webots/actions/runs/4934984486?pr=6061
Then you can run it from the icon and report what do you get in the Webots console.

CurryYH, 2023-05-10:
> That's because you would need to call `webots-bin.exe` instead of `webotsw.exe`. However, given that it can be tricky to setup correctly the environment manually to execute the binary directly, I [...]

**Here is the output from Webots console:**

[console_output.txt](https://github.com/cyberbotics/webots/files/11442374/console_output.txt)

stefaniapedrazzi, 2023-05-11:
Ok! So finally it seems that the issue is caused by the user name containing the `'` character.
I'm fixing it in #6131.

stefaniapedrazzi, 2023-05-11:
@CurryYH the fixes has just been merged and will be included in the nightly build of Webots R2023a revision 1 starting from tomorrow (11_5_1023):
https://github.com/cyberbotics/webots/tags

closed 2023-05-11 by stefaniapedrazzi

CurryYH, 2023-05-11:
> @CurryYH the fixes has just been merged and will be included in the nightly build of Webots R2023a revision 1 starting from tomorrow (11_5_1023): https://github.com/cyberbotics/webots/tags

That's good and thank you very much, I will download the nightly build version tomorrow to check whether it works or not.

CurryYH, 2023-05-11:
> @CurryYH the fixes has just been merged and will be included in the nightly build of Webots R2023a revision 1 starting from tomorrow (11_5_1023): https://github.com/cyberbotics/webots/tags

That's great! Thanks so much. I will download the nightly build version tomorrow to verify if it functions properly.

PR #6131 (2023-05-11): Fix errors loading template PROTO nodes
Fix #6060:
if a path used in the Javascript (or Lua) template file contains the `'` character, the code becomes invalid and it is not possible to load the template PROTO.
All the paths need to be correctly escaped before writing them in the template file.