cyberbotics/webots #6072: Joint with SolidReference as endPoint inside mechanical loop gets unlinked when there's a SliderJoint in the loop

IDavGal, 2023-04-14:
**Describe the Bug**
A joint where the endPoint is a SolidReference, inside a mechanical loop that has a slider joint in it, acts as if there was joint at all.

**Steps to Reproduce**
1. Open the attached project file [[example_project.zip](https://github.com/cyberbotics/webots/files/11236250/example_project.zip)]
2. Wait for the controller to move the LinearMotor up and down
3. Corroborate that the corresponding solids are not attached by the joint where the SolidReference is specified

**Expected behavior**
In the example given, 2 robots are spawned, each with a different joint having a SolidReference as endPoint.
In each case, both upper and lower beams start points are connected to the base_link with a hingeJoint, while the sliderJoint is also connected to each beam's outer end via hingeJoints.
I would expect the upper beam to rotate around the joint that links them to the base link when the sliderJoint extends. Instead I see that the beam moves as if no joint was specified

**Screenshots**
Initial state:
[image]

Extended sliderJoint:
[image]

**System**
 - Operating System: [ Linux Ubuntu 20.04]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 3050 4 GB]

**Additional context**
- I think this may be related to this [older issue](https://github.com/cyberbotics/webots/issues/3163)
- A similar mechanism with a solid instead of a SliderJoint works as expected
- A diagram is provided to better illustrate the mechanical loop
[image]

IDavGal, 2023-05-04:
Hi! Any update on this?

stefaniapedrazzi, 2023-05-08:
I didn't have the time to inspect the issue yet.

From my first quick inspection it seems that I can get the exact same behavior replacing the `SliderJoint` with a `HingeJoint.
Then, I see that in your example all the Solid nodes are kinematic.
Do you get the same behavior properly setting the boundingObject and physics nodes for the Solid in your robot?