cyberbotics/webots #6078: Build R2023a and all Nightly Build 2023: WeBots doesn't start (dll files were not found on windows 11)

ali1hammoud, 2023-04-16:
After installing webots 2023 and trying all versions, for Windows, in the form of EXE installer,
WeBots doesn't start because of a missing DLL, libgcc_s_seh-1.dll, libstdc++-6.dll, libwinpthread-1.dll

Operating System: Windows 10

BenjaminDeleze, 2023-04-17:
I tried to install R2023a and the latest Nightly Build both on Windows 10 and Windows 11 and got no problem.
It may be possible that the problem is coming from your system.

ali1hammoud, 2023-04-17:
Thank you for your answer.
I downloaded these file from internet ( libgcc_s_seh-1.dll, libstdc++-6.dll, libwinpthread-1.dll), and added them to this path:
C:\Program Files\Webots\msys64\mingw64\bin

but I got these errors:
[image]

[image]

[image]

[image]

are there any ideas where the problem could be?

BenjaminDeleze, 2023-04-18:
@omichel do you have an idea?

omichel, 2023-04-18:
No idea. But it is probably a bad idea to download libraries from the Internet as they are likely not the same version as the ones supposed to be included within Webots. These 3 libraries are normally installed by Webots inside the `C:\Program Files\Webots\msys64\mingw64\bin\cpp` folder. Don't you have them here?

ali1hammoud, 2023-04-18:
Yes these files inside the `C:\Program Files\Webots\msys64\mingw64\bin\cpp` folder, but I don't know why I got that error.

I copied the files to this folder:   `C:\Program Files\Webots\msys64\mingw64\bin`, and now I have this error:
[image]

omichel, 2023-04-18:
Did you try to launch Webots from the icon (instead of launching it from the DOS console)?

ali1hammoud, 2023-04-18:
Yes, and nothing is showing.

in task manager, I see that webots start for few seconds then the process end immediately without any GUI.

omichel, 2023-04-18:
I have no more idea... That could be a problem with your local firewall...