cyberbotics/webots #6126: memory leaks with NAO robot using RL

hengyue-dunhuang, 2023-05-09:
**Describe the Bug**
It seems that memory leaks happend when I trained my NAO robot with REINFORCEMENT LEARNING.

**Screenshots**
[image]
after few minutes 
[image]

**System**
 -Windows 10 
- Graphics Card: NVIDIA 1660super 6 GB

**Additional Contents**
This is my world
[image]
This is NAO 
[image]

ygoumaz, 2023-05-09:
Could you please provide some additional details so that I can investigate the source of the leak?
- What controller are you using?
- Are you using an external script to run multiple scenarios for your reinforcement learning? If yes, what is it doing?
- Can you detail a little bit more the process of your algorithm: is it starting multiple instances of Webots, reloading the world multiple times, etc?

If you have access to a Linux machine, I suggest you to run your project there and use the [Heaptrack](https://github.com/cyberbotics/webots/wiki/Heaptrack) tool which is very intuitive to find memory leak sources. Otherwise, if possible, you can also provide your project and I can have a look myself.

hengyue-dunhuang, 2023-05-09:
Thanks for your reply, I will do the same thing in my ubuntu18.04 with the help of **Heaptrack**.

ShuffleWire, 2026-04-05:
@hengyue-dunhuang Have you made any progress, did you need any help ? Else could we/you close to issue ?