cyberbotics/webots #6137: Robot stops at very low velocity

ThorbenCarl, 2023-05-13:
**Bug? Description**
Hello everyone,

we have built a robot (track.wbt) and would like to make it drive straight ahead with a very low speed (my_controller.cpp). `maxVelocity` of the two LinearMotors of the tracks are limited to 0.0036 m/s.
If we enter a speed of `trackLeft->setVelocity(0.0036); trackRight->setVelocity(0.0036);` for both drives, the robot drives for approx 1s and then moves on the spot. The animations are still moving and we also get a speed, but the position doesn't change.
The problem does not occur if we set the `maxVelocity` to 0.02 m/s and control both drives at 0.02 m/s.
The problem also does not occur if the difference between the two drives is large enough, e.g. left -0.0036 and right 0.0036 or left 0.0036 and right 0.0025.

We are currently calling setVelocity cyclically to allow the robot to drive straight, but this is rather unpleasant for performance reasons.

What are we doing wrong? Is it a friction or force problem, but why the robot not stops and return a velocity of 0? We have already reduced the size of the robot's shield to make sure it doesn't touch the ground, but it had no effect. Can you reproduce the problem? Neither Google nor ChatGPT could help us :D

**Steps to Reproduce**
[zumo_sim.zip](https://github.com/cyberbotics/webots/files/11469289/zumo_sim.zip)
1. Load World/Robot
2. Choose my_Controller for the Robot(Zumo)
3. Start simulation 
4. Change Velocities 
5. and see Robot stopping or not

**Expected behavior**
Robot should drive straight.

**System**
@ThorbenCarl
 - Operating System: macOS Ventura 13.3.1 (a) (22E772610a)
 - Graphics Card: Intel Iris Plus Graphics 655 1536 MB
- (MacBook Pro, 2,3 GHz Quad-Core Intel Core i5, 8 GB 2133 MHz LPDDR3)
- Webots 2023a

dsiev, 2023-05-13:
Same here, no idea why.

**System:**
- OS: Windows 10 Home, Version 22H2 (Build 19045.2965)
- Graphics Card: NVIDIA Geforce GTX 1050
- Processor: Intel(R) Core(TM) i5-7300HQ CPU @ 2.50GHz x64
- Webots R2023a

omichel, 2023-05-15:
Because the motion is so slow and the collisions are not changing much at such a low speed, Webots considers the robot is not moving and thus disables the corresponding bodies, causing the motion to completely stop. To fix this, you can simply decrease the values of `WorldInfo.physicsDisableLinearThreshold` and `WorldInfo.physicsDisableAngularThreshold`. Setting them to 0.001 (instead of their default value of 0.01) should fix the problem. Alternatively, setting `WorldInfo.physicsDisableTime` to -1 will also fix the problem in a less optimal way. See the documentation about these parameters [here](https://cyberbotics.com/doc/reference/worldinfo).

Please let me know if that resolves the issue for you guys.