cyberbotics/webots #6140: Fail to start on Windows

SpyrosKou, 2023-05-15:
**Describe the Bug**
After installing the [webots-R2023a_setup.exe](https://github.com/cyberbotics/webots/releases/download/R2023a/webots-R2023a_setup.exe) I get a small pop up window saying:

> The application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

I see that a similar issue [has been also encountered in the past.](https://github.com/cyberbotics/webots/issues/33) 

**Steps to Reproduce**
1. [Install the R2023a version](https://github.com/cyberbotics/webots/releases/download/R2023a/webots-R2023a_setup.exe)
2. Start the installed version
3. After uninstalling and reinstalling or reinstalling directly, the same behavior is observed.

**Expected behavior**
Webots should have started after installation.

**Screenshots**
[image]

**System**
 - Operating System: Microsoft 11 Pro
 - Graphics Card: NVIDIA RTX A3000 Laptop GPU

SpyrosKou, 2023-05-15:
@omichel , thank you for your answer.
I  tried [nightly_12_5_2023/webots-R2023a-rev1](https://github.com/cyberbotics/webots/releases/download/nightly_12_5_2023/webots-R2023a-rev1_setup.exe) and the result is the same.

I also did a reinstall of the [nightly_12_5_2023/webots-R2023a-rev1](https://github.com/cyberbotics/webots/releases/download/nightly_12_5_2023/webots-R2023a-rev1_setup.exe) version.

omichel, 2023-05-16:
That is strange as you are the only one user reporting such a problem. It could come from a conflict with another application installed on your computer. Do you have some other Qt6 libraries installed somewhere on your computer?
To investigate the problem, I would recommend you to install [MSYS2](https://www.msys2.org) on your system, and from the MSYS2 console, type the following:
```bash
$ cd /C/Program\ Files/Webots/msys64/mingw64/bin/
$ ldd webots-bin.exe
```
On my computer I am getting the following output:
```
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffc7f9f0000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffc7e750000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffc7d1f0000)
        ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ffc7e960000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ffc7dd80000)
        sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ffc7e6a0000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ffc7da20000)
        ole32.dll => /c/WINDOWS/System32/ole32.dll (0x7ffc7e500000)
        d3d9.dll => /c/WINDOWS/SYSTEM32/d3d9.dll (0x7ffc65ed0000)
        msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ffc7d620000)
        GLU32.dll => /c/WINDOWS/SYSTEM32/GLU32.dll (0x7ffc741f0000)
        gdi32.dll => /c/WINDOWS/System32/gdi32.dll (0x7ffc7de30000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ffc7cf30000)
        libgcc_s_seh-1.dll => /mingw64/bin/libgcc_s_seh-1.dll (0x7ffc6f740000)
        USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ffc7dbc0000)
        win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ffc7d1c0000)
        combase.dll => /c/WINDOWS/System32/combase.dll (0x7ffc7dff0000)
        IPHLPAPI.DLL => /c/WINDOWS/SYSTEM32/IPHLPAPI.DLL (0x7ffc7b9b0000)
        gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ffc7ce10000)
        libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x7ffc71620000)
        OPENGL32.dll => /c/WINDOWS/SYSTEM32/OPENGL32.dll (0x7ffc34c10000)
        libopenal-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/libopenal-1.dll (0x7ffc32a40000)
        SHELL32.dll => /c/WINDOWS/System32/SHELL32.dll (0x7ffc7f1b0000)
        libassimp-5.dll => /c/Program Files/Webots/msys64/mingw64/bin/libassimp-5.dll (0x7ffc0b370000)
        libfreetype-6.dll => /c/Program Files/Webots/msys64/mingw64/bin/libfreetype-6.dll (0x7ffc6b9e0000)
        libpico.dll => /c/Program Files/Webots/msys64/mingw64/bin/libpico.dll (0x7ffc69b70000)
        lua52.dll => /c/Program Files/Webots/msys64/mingw64/bin/lua52.dll (0x7ffc331d0000)
        libstdc++-6.dll => /mingw64/bin/libstdc++-6.dll (0x7ffc27b70000)
        ode.dll => /c/Program Files/Webots/msys64/mingw64/bin/ode.dll (0x7ffc32710000)
        OIS.dll => /c/Program Files/Webots/msys64/mingw64/bin/OIS.dll (0x6e3c0000)
        openvr_api.dll => /c/Program Files/Webots/msys64/mingw64/bin/openvr_api.dll (0x7ffc33180000)
        OLEAUT32.dll => /c/WINDOWS/System32/OLEAUT32.dll (0x7ffc7e880000)
        Qt6Core.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Core.dll (0x7ffc0adb0000)
        Qt6Gui.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Gui.dll (0x7ffc0a5f0000)
        WS2_32.dll => /c/WINDOWS/System32/WS2_32.dll (0x7ffc7df70000)
        Qt6Network.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Network.dll (0x7ffc23f80000)
        Qt6OpenGL.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6OpenGL.dll (0x7ffc32e70000)
        Qt6OpenGLWidgets.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6OpenGLWidgets.dll (0x7ffc6fe20000)
        Qt6PrintSupport.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6PrintSupport.dll (0x7ffc33110000)
        comdlg32.dll => /c/WINDOWS/System32/comdlg32.dll (0x7ffc7d860000)
        dwmapi.dll => /c/WINDOWS/SYSTEM32/dwmapi.dll (0x7ffc79fd0000)
        Qt6WebSockets.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6WebSockets.dll (0x7ffc42e90000)
        Qt6Qml.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Qml.dll (0x7ffc0a100000)
        shcore.dll => /c/WINDOWS/System32/shcore.dll (0x7ffc7de70000)
        Qt6Xml.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Xml.dll (0x7ffc6f710000)
        Qt6Widgets.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Widgets.dll (0x7ffc094f0000)
        SHLWAPI.dll => /c/WINDOWS/System32/SHLWAPI.dll (0x7ffc7e820000)
        libminizip-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/libminizip-1.dll (0x7ffc6b9b0000)
        libbz2-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/libbz2-1.dll (0x7ffc6f9d0000)
        WINMM.dll => /c/WINDOWS/SYSTEM32/WINMM.dll (0x7ffc7aa00000)
        libbrotlidec.dll => /c/Program Files/Webots/msys64/mingw64/bin/libbrotlidec.dll (0x7ffc6f600000)
        libpng16-16.dll => /c/Program Files/Webots/msys64/mingw64/bin/libpng16-16.dll (0x7ffc32e20000)
        libharfbuzz-0.dll => /c/Program Files/Webots/msys64/mingw64/bin/libharfbuzz-0.dll (0x7ffc29460000)
        zlib1.dll => /c/Program Files/Webots/msys64/mingw64/bin/zlib1.dll (0x7ffc6a2d0000)
        zlib1.dll => /c/Program Files/Webots/msys64/mingw64/bin/zlib1.dll (0x1fd98350000)
        XINPUT9_1_0.dll => /c/Program Files/Webots/msys64/mingw64/bin/XINPUT9_1_0.dll (0x7ffc79260000)
        AUTHZ.dll => /c/WINDOWS/SYSTEM32/AUTHZ.dll (0x7ffc7bae0000)
        DINPUT8.dll => /c/WINDOWS/SYSTEM32/DINPUT8.dll (0x7ffc32dd0000)
        MPR.dll => /c/WINDOWS/SYSTEM32/MPR.dll (0x7ffc69d50000)
[... cut ...]
        libpcre2-8-0.dll => /c/Program Files/Webots/msys64/mingw64/bin/libpcre2-8-0.dll (0x7ffc297c0000)
        libiconv-2.dll => /c/Program Files/Webots/msys64/mingw64/bin/libiconv-2.dll (0x7ffc0ea70000)
        CRYPTBASE.DLL => /c/WINDOWS/SYSTEM32/CRYPTBASE.DLL (0x7ffc7c640000)
        SSPICLI.DLL => /c/WINDOWS/SYSTEM32/SSPICLI.DLL (0x7ffc7c180000)
        NETUTILS.DLL => /c/WINDOWS/SYSTEM32/NETUTILS.DLL (0x7ffc7b8f0000)
        SRVCLI.DLL => /c/WINDOWS/SYSTEM32/SRVCLI.DLL (0x7ffc6bc90000)
```
I suspect you will get something different which will hopefully highlight a possible DLL conflict.

SpyrosKou, 2023-05-16:
Thank you @omichel for your response.

Here is the output  of [MSYS2](https://www.msys2.org/).
```
$ cd /C/Program\ Files/Webots/msys64/mingw64/bin/
$ ldd webots-bin.exe

```

```
      ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffc1f810000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffc1e740000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffc1cf60000)
        ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ffc1e950000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ffc1e610000)
        sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ffc1e0d0000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ffc1d610000)
        ole32.dll => /c/WINDOWS/System32/ole32.dll (0x7ffc1e2b0000)
        d3d9.dll => /c/WINDOWS/SYSTEM32/d3d9.dll (0x7ffc01900000)
        msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ffc1cc30000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ffc1d4c0000)
        GLU32.dll => /c/WINDOWS/SYSTEM32/GLU32.dll (0x7ffbfc5d0000)
        gdi32.dll => /c/WINDOWS/System32/gdi32.dll (0x7ffc1f050000)
        USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ffc1e450000)
        user32.dll => /c/Windows/System32/user32.dll (0x2b111f00000)
        win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ffc1d5e0000)
        win32u.dll => /c/Windows/System32/win32u.dll (0x2b111f00000)
        combase.dll => /c/WINDOWS/System32/combase.dll (0x7ffc1ea80000)
        gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ffc1ccd0000)
        libgcc_s_seh-1.dll => not found
        IPHLPAPI.DLL => /c/WINDOWS/SYSTEM32/IPHLPAPI.DLL (?)
        libopenal-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/libopenal-1.dll (?)
        OPENGL32.dll => /c/WINDOWS/SYSTEM32/OPENGL32.dll (?)
        libstdc++-6.dll => not found
        libassimp-5.dll => /c/Program Files/Webots/msys64/mingw64/bin/libassimp-5.dll (?)
        libfreetype-6.dll => /c/Program Files/Webots/msys64/mingw64/bin/libfreetype-6.dll (?)
        libpico.dll => /c/Program Files/Webots/msys64/mingw64/bin/libpico.dll (?)
        lua52.dll => /c/Program Files/Webots/msys64/mingw64/bin/lua52.dll (?)
        ode.dll => /c/Program Files/Webots/msys64/mingw64/bin/ode.dll (?)
        OIS.dll => /c/Program Files/Webots/msys64/mingw64/bin/OIS.dll (?)
        openvr_api.dll => /c/Program Files/Webots/msys64/mingw64/bin/openvr_api.dll (?)
        Qt6Core.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Core.dll (?)
        Qt6Gui.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Gui.dll (?)
        Qt6Network.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Network.dll (?)
        Qt6OpenGL.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6OpenGL.dll (?)
        Qt6OpenGLWidgets.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6OpenGLWidgets.dll (?)
        Qt6PrintSupport.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6PrintSupport.dll (?)
        Qt6Qml.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Qml.dll (?)
        Qt6WebSockets.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6WebSockets.dll (?)
        Qt6Widgets.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Widgets.dll (?)
        Qt6Xml.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Xml.dll (?)
```

omichel, 2023-05-16:
It seems that Webots is unable to find libraries inside your `C:\Program Files\Webots\msys64\mingw64\bin\cpp` folder. Can you check that this folder exists and contains the missing libraries?
```bash
$ ls -l /C/Program\ Files/Webots/msys64/mingw64/bin/cpp
total 2148
-rwxr-xr-x 1 Olivi Olivi  107187 Jan 23 07:22 libgcc_s_seh-1.dll
-rwxr-xr-x 1 Olivi Olivi 2025890 Jan 23 07:22 libstdc++-6.dll
-rwxr-xr-x 1 Olivi Olivi   60356 Mar  3 10:44 libwinpthread-1.dll
```
If so, can you add the path to it to your `PATH` environment variable and try again:
```bash
$ export PATH=$PATH:/C/Program\ Files/Webots/msys64/mingw64/bin/cpp
$ ldd /C/Program\ Files/Webots/msys64/mingw64/bin/webots-bin.exe
```

SpyrosKou, 2023-05-16:
The results show that the files exist.
`ls -l /C/Program\ Files/Webots/msys64/mingw64/bin/cpp`
```
total 2148
-rwxr-xr-x 1 auser auser  107699 Oct 13  2022 libgcc_s_seh-1.dll
-rwxr-xr-x 1 auser auser 2025245 Oct 13  2022 libstdc++-6.dll
-rwxr-xr-x 1 auser auser 58109 Oct 15  2022 libwinpthread-1.dll
```

Afterwards I add the folder to the path and run the command:
```
$ ldd /C/Program\ Files/Webots/msys64/mingw64/bin/webots-bin.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffc1f810000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffc1e740000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffc1cf60000)
        ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ffc1e950000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ffc1e610000)
        sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ffc1e0d0000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ffc1d610000)
        ole32.dll => /c/WINDOWS/System32/ole32.dll (0x7ffc1e2b0000)
        msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ffc1cc30000)
        d3d9.dll => /c/WINDOWS/SYSTEM32/d3d9.dll (0x7ffc01900000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ffc1d4c0000)
        gdi32.dll => /c/WINDOWS/System32/gdi32.dll (0x7ffc1f050000)
        GLU32.dll => /c/WINDOWS/SYSTEM32/GLU32.dll (0x7ffbfc5d0000)
        win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ffc1d5e0000)
        user32.dll => /c/Windows/System32/user32.dll (0x1610c790000)
        gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ffc1ccd0000)
        USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ffc1e450000)
        combase.dll => /c/WINDOWS/System32/combase.dll (0x7ffc1ea80000)
        libgcc_s_seh-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/cpp/libgcc_s_seh-1.dll (0x7ffbdc870000)
        IPHLPAPI.DLL => /c/WINDOWS/SYSTEM32/IPHLPAPI.DLL (0x7ffc1b790000)
        OPENGL32.dll => /c/WINDOWS/SYSTEM32/OPENGL32.dll (0x7ffbfa3b0000)
        dwmapi.dll => /c/WINDOWS/SYSTEM32/dwmapi.dll (0x7ffc17640000)
        libwinpthread-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/cpp/libwinpthread-1.dll (0x7ffbdd3e0000)
        libstdc++-6.dll => /c/Program Files/Webots/msys64/mingw64/bin/cpp/libstdc++-6.dll (0x7ffb5b7b0000)
        libpico.dll => /c/Program Files/Webots/msys64/mingw64/bin/libpico.dll (0x7ffbd0be0000)
        libfreetype-6.dll => /c/Program Files/Webots/msys64/mingw64/bin/libfreetype-6.dll (0x7ffbae320000)
        libassimp-5.dll => /c/Program Files/Webots/msys64/mingw64/bin/libassimp-5.dll (0x7ffb2ff40000)
        libopenal-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/libopenal-1.dll (0x7ffb901e0000)
        SHELL32.dll => /c/WINDOWS/System32/SHELL32.dll (0x7ffc1d730000)
        openvr_api.dll => /c/Program Files/Webots/msys64/mingw64/bin/openvr_api.dll (0x7ffbda4b0000)
        lua52.dll => /c/Program Files/Webots/msys64/mingw64/bin/lua52.dll (0x7ffbd9640000)
        ode.dll => /c/Program Files/Webots/msys64/mingw64/bin/ode.dll (0x7ffb79f10000)
        OIS.dll => /c/Program Files/Webots/msys64/mingw64/bin/OIS.dll (0x6e3c0000)
        OLEAUT32.dll => /c/WINDOWS/System32/OLEAUT32.dll (0x7ffc1e870000)
        Qt6OpenGL.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6OpenGL.dll (0x7ffbca960000)
        Qt6Network.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Network.dll (0x7ffb5b5e0000)
        WS2_32.dll => /c/WINDOWS/System32/WS2_32.dll (0x7ffc1efd0000)
        Qt6OpenGLWidgets.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6OpenGLWidgets.dll (0x7ffbdb630000)
        Qt6PrintSupport.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6PrintSupport.dll (0x7ffbce780000)
        comdlg32.dll => /c/WINDOWS/System32/comdlg32.dll (0x7ffc1f0e0000)
        shcore.dll => /c/WINDOWS/System32/shcore.dll (0x7ffc1f1e0000)
        SHLWAPI.dll => /c/WINDOWS/System32/SHLWAPI.dll (0x7ffc1e810000)
        Qt6Core.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Core.dll (0x7ffb2f980000)
        Qt6WebSockets.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6WebSockets.dll (0x7ffbda9e0000)
        Qt6Qml.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Qml.dll (0x7ffb2f490000)
        Qt6Xml.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Xml.dll (0x7ffbd9f50000)
        libbz2-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/libbz2-1.dll (0x7ffbd9720000)
        Qt6Gui.dll => /c/Program Files/Webots/msys64/mingw64/bin/Qt6Gui.dll (0x7ffb2ecd0000)
        libbrotlidec.dll => /c/Program Files/Webots/msys64/mingw64/bin/libbrotlidec.dll (0x7ffbd8930000)
        libharfbuzz-0.dll => /c/Program Files/Webots/msys64/mingw64/bin/libharfbuzz-0.dll (0x7ffba8370000)
        libpng16-16.dll => /c/Program Files/Webots/msys64/mingw64/bin/libpng16-16.dll (0x7ffbd0b90000)
        zlib1.dll => /c/Program Files/Webots/msys64/mingw64/bin/zlib1.dll (0x7ffbd51a0000)
        zlib1.dll => /c/Program Files/Webots/msys64/mingw64/bin/zlib1.dll (0x1610c790000)
        WINMM.dll => /c/WINDOWS/SYSTEM32/WINMM.dll (0x7ffc10920000)
        DINPUT8.dll => /c/WINDOWS/SYSTEM32/DINPUT8.dll (0x7ffbce5c0000)
        libminizip-1.dll => /c/Program Files/Webots/msys64/mingw64/bin/libminizip-1.dll (0x7ffbd4db0000)
        DNSAPI.dll => /c/WINDOWS/SYSTEM32/DNSAPI.dll (0x7ffc1b7c0000)
        Secur32.dll => /c/WINDOWS/SYSTEM32/Secur32.dll (0x7ffbf7540000)
[... 39 lines cut ...]
The command  was `$ ./webotsw.exe`
The command return the following, extra information.
```
[... cut ...]
Warning: In C:/Qt/Qt5.12.12/5.12.12/msvc2017_64/plugins/platforms/qwindowsd.dll:
  Plugin uses incompatible Qt library (5.12.0) [debug]
Warning: Could not find the Qt platform plugin "windows" in "C:\Qt\Qt5.12.12\5.12.12\msvc2017_64\plugins\platforms"
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```

There is a warning for another QT instance.
I run the same command again without the updated path, and returns the same.

omichel, 2023-05-16:
All right, so I believe the problem is that Webots tries to load the plugins from your `C:\Qt\Qt5.12.12` folder whereas it should use the one from `Qt6` embedded with Webots in `C:\Program Files\Webots\msys64\mingw64\share\qt6\plugins`. It could be that you have some environment variable set pointing to this Qt5 folder, like `QT_PLUGIN_PATH`. Can you test whether this environment variable is set?
```
$ printenv | grep QT
```

SpyrosKou, 2023-05-16:
Webots run successfully!
The problem was the existence of two environment variables.
I used the following commands to `unset` them in the `MSYS` terminal and started webots from there.
```
$ unset Qt5_DIR
$ unset QT_QPA_PLATFORM_PLUGIN_PATH
```
Thank you for your support!

closed 2023-05-16 by SpyrosKou

omichel, 2023-05-17:
Nice to read!
I believe that you should remove the `QT_QPA_PLATFORM_PLUGIN_PATH` environment variable from your system settings as it may conflict with other applications as well. And that will allow you to start Webots from the Windows GUI. Normally, Qt5 should be able to find its plugin path from the `Qt5_DIR` variable.

SpyrosKou, 2023-05-17:
Thank you for the advice.
I confirm that after removing the `QT_QPA_PLATFORM_PLUGIN_PATH` , Webots start normally from Windows UI.

miquelmassot, 2023-06-27:
I've hit the same issue - I have Qt5 installed on my Windows machine, which populated the environmental variables `QT_QPA_PLATFORM_PLUGIN_PATH` and `Qt5_DIR`.

omichel, 2023-06-27:
I tried to fix the problem in #6258. Can you try it and confirm it resolves the problem for you?
If you don't want to compile Webots from the source, you can simply replace the Webots launcher from [Webots R2023b](https://github.com/cyberbotics/webots/releases/tag/R2023b) by the one in this [webots.zip](https://github.com/cyberbotics/webots/files/11880621/webots.zip) file: download and unzip the file, then copy it in your Webots installation folder, e.g., in `C:\Program Files\Webots\msys64\mingw64\bin\` as a replacement of the original `webots.exe` (you will need administrator rights). Let me know if that works for you.