cyberbotics/webots #6174: Supervisor return None object on 2023 version

kadirou31, 2023-06-06:
Hi,
i'm using Webots 2023a under Ubuntu 22 LST, I'm trying to use Supervisor based on the tutorial code, I'm getting this output: 
---------------------------------------------------------------------------------------------------------------------------------------------------
traceback (most recent call last):
  File "C:\Users\Admin\Documents\my_project\controllers\my_controller_sup\my_controller_sup.py", line 10, in <module>
    trans_field = robot_node.getField("translation")
AttributeError: 'NoneType' object has no attribute 'getField'
WARNING: 'my_controller_sup' controller exited with status: 1.
---------------------------------------------------------------------------------------------------------------------------------------------------
i noticed that robot_node is None !!
the used code is : 
from math import sqrt
from controller import Supervisor

TIME_STEP = 32

supervisor = Supervisor()

# get handle to robot's translation field
robot_node = supervisor.getFromDef("e-puck")
trans_field = robot_node.getField("translation")

for a in range(0, 25):
    for b in range(0, 33):
        # evaluate robot during 60 seconds (simulation time)
        t = supervisor.getTime()
        while supervisor.getTime() - t < 60:

            # perform robot control according to a, b
            # (and possibly t) parameters.

            # controller termination
            if supervisor.step(TIME_STEP) == -1:
                quit()

        # compute travelled distance
        values = trans_field.getSFVec3f()
        dist = sqrt(values[0] * values[0] + values[2] * values[2])
        print("a=%d, b=%d -> dist=%g" % (a, b, dist))

        # reset robot position and physics
        INITIAL = [0, 0.5, 0]
        trans_field.setSFVec3f(INITIAL)
        robot_node.resetPhysics()

BenjaminDeleze, 2023-06-07:
Hello,
It works well on my side.

Please check:

- That your robot's DEF is `e-puck`
- That your robot is a supervisor

kadirou31, 2023-06-07:
Hi, I tried it on Windows and Linux, supervisor is enabled on the robot. Could you please help to resolve this issue.

BenjaminDeleze, 2023-06-07:
Can you post your world file (`.wbt`)?

kadirou31, 2023-06-07:
[my_supervisor.zip](https://github.com/cyberbotics/webots/files/11674499/my_supervisor.zip)

BenjaminDeleze, 2023-06-07:
There is no DEF in your world so it is normal that the `getFromDef` function returns `null`

closed 2023-06-07 by BenjaminDeleze

kadirou31, 2023-06-07:
how to add DEF to the world

skpawar1305, 2025-02-12:
getField is returning None always in 2025a for me
is it the same for you?

JoelRamosBeltran, 2025-03-31:
> getField is returning None always in 2025a for me is it the same for you?

I have the same problem. And then, the command 
`self.__insertion_node_place.importMFNodeFromString(-1, object_string)`
returns 
"[ros2_supervisor.py-2] AttributeError: 'NoneType' object has no attribute 'importMFNodeFromString'"

I am using the /opt/ros/humble/lib/webots_ros2_driver/ros2_supervisor.py with a launcher:

`    webots = WebotsLauncher(
        world=os.path.join(package_dir, 'worlds', 'Prueba_rbrobout_webots.wbt'),
        mode=mode,
        ros2_supervisor=True
    )`

And then I tried to spawn a robot from a PROTO with:

ros2 service call /Ros2Supervisor/spawn_node_from_string webots_ros2_msgs/srv/SpawnNodeFromString "data: Robot { name \"rbrobout\" }"

My webots world is:

```
#VRML_SIM R2025a utf8

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2025a/projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2025a/projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2025a/projects/objects/floors/protos/RectangleArena.proto"
IMPORTABLE EXTERNPROTO "../protos/rbrobout.proto"

WorldInfo {
}
Viewpoint {
  orientation 0.10191405677525116 -0.18341781994458842 -0.9777379139413515 5.237674100104831
  position -1.9942878694861297 -4.803766957859288 0.4791667009234102
}
TexturedBackground {
}
TexturedBackgroundLight {
}
RectangleArena {
  translation 0 0 -0.64
  floorSize 10 10
  floorTileSize 1 1
  wallHeight 0.3
}
```

skpawar1305, 2025-03-31:
> I have the same problem. And then, the command
> `self.__insertion_node_place.importMFNodeFromString(-1, object_string)`
> returns
> "[ros2_supervisor.py-2] AttributeError: 'NoneType' object has no [...]

for me everything is fine now, check if you're using updated repo

JoelRamosBeltran, 2025-03-31:
> for me everything is fine now, check if you're using updated repo

I am using Webots 2025a with Ros2 Humble. I think that it could be a problem. I don't know if the updated repo is compatible with Humble. The last version with "sudo apt-get install ros-humble-webots-ros2" is "2023.1.3-1jammy.20250207.204829"