cyberbotics/webots #6194: Inconsistent behaviour of lidar in python

miquelmassot, 2023-06-13:
Hi,

I am trying to code a simple driver in Python using a lidar. So far, I've tried the range image and the point cloud outputs but neither of them work - the console reports that "The process crashed sometime after starting successfully". 

The code to test this is attached: 
[webots.zip](https://github.com/cyberbotics/webots/files/11730873/webots.zip)

Also, the documentation and the API do not match here:
https://github.com/cyberbotics/webots/blob/cb1af9e3bbe43a138b70fe67552cfcc8c7fe1e11/lib/controller/python/controller/lidar.py#L142
https://cyberbotics.com/doc/reference/lidar?tab-language=python#wb_lidar_get_point_cloud

where the argument data_type in getPointCloud() is missing in the code.

omichel, 2023-06-13:
Which version are you using? I would recommend to try the latest nightly build of Webots R2022a-rev1 from [here](https://github.com/cyberbotics/webots/releases) as it contains fixes for many bugs and inconsistencies in the Python API.

omichel, 2023-06-13:
I modified the [python lidar example](https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/controllers/lidar/lidar.py) and added the following code from [line 47](https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/controllers/lidar/lidar.py#L47):

```python
            point_cloud = self.lidar.getPointCloud()
            print(str(type(point_cloud)) + ' of ' + str(type(point_cloud[0])) + ' value[0].x = ' + str(point_cloud[0].x))
            range_image = self.lidar.getRangeImage()
            print(str(type(range_image)) + ' of ' + str(type(range_image[0])) + ' value[0] = ' + str(range_image[0]))      
```
When I run it in [lidar.wbt](https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/worlds/lidar.wbt) after removing the C binary controller from the [controller folder](https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/controllers/lidar/), I am getting the following output:
```
<class 'list'> of <class 'controller.lidar_point.LidarPoint'> value[0].x = 2.1800849437713623
<class 'list'> of <class 'float'> value[0] = 2.1806445121765137
<class 'list'> of <class 'controller.lidar_point.LidarPoint'> value[0].x = 2.1800849437713623
<class 'list'> of <class 'float'> value[0] = 2.1806445121765137
<class 'list'> of <class 'controller.lidar_point.LidarPoint'> value[0].x = 2.1800849437713623
<class 'list'> of <class 'float'> value[0] = 2.1806445121765137
<class 'list'> of <class 'controller.lidar_point.LidarPoint'> value[0].x = 2.1800849437713623
<class 'list'> of <class 'float'> value[0] = 2.1806445121765137
```
I did not observe any crash.
Can you try the same on your side and report if you are getting the same results as I do?

omichel, 2023-06-13:
I don't have any clue. I would recommend you to modify the working example step-by-step until it fully resembles your world and controller. At some point, it should break and you will understand what is the cause of the problem. Once you identified it, please post the information here so that we can help with this case.

closed 2026-11-11 by omichel