cyberbotics/webots #6211: Classic theme issues

stefaniapedrazzi, 2023-06-15:
I'm testing the snap package on Ubuntu 22.04
* the dialog to choose the editor  font from the Preferences dialog is broken.
    When clicking on the button to open it, the UI becomes gray (because it is a modal dialog that should block the rest of the application) but nothing is displayed.

BenjaminDeleze, 2023-06-15:
This one?
[image]

Strange, it works well on snap ubuntu 20.04

stefaniapedrazzi, 2023-06-15:
Yes. It doesn't show up for me.

stefaniapedrazzi, 2023-06-15:
I also checked on my development environment (same system) and works fine.
The issues seems only with the snap installation.

BenjaminDeleze, 2023-06-15:
Very strange as it is working for snap on ubuntu 22.
Does it happens only for the classic theme?

stefaniapedrazzi, 2023-06-15:
After i opened it on the development environment it works also from the snap.
I will check on a clean VM to see if I can reproduce it.

The second issue is with the video parameter checkbox is reproducible consistently.

BenjaminDeleze, 2023-06-15:
Yes I have it too

PR #6219 (2023-06-15): Fix video dialog QCheckBox
**Description**
Fix #6211 
I checked all the QCheckBoxes and this was the only one that was not working.
closed 2023-06-15 by BenjaminDeleze