cyberbotics/webots #6215: Backward mechanism

BenjaminDeleze, 2023-06-15:
**Describe the Bug**
The backward mechanism is working quite well for `Pose`/`Transform`.
However there is some cases where some objects are not displayed correctly:
[image]

Is it an acceptable degree of backward compatibility?

omichel, 2023-06-15:
That's odd. Why are we getting this?

BenjaminDeleze, 2023-06-15:
I still need to investigate

BenjaminDeleze, 2023-06-15:
It happens because, as I am loading a world from R2023a, it loads the `Toilet.proto` from R2023a that is something like:
```
Solid {
    translation IS translation
    rotation IS rotation
    scale 0.2 0.2 0.2
    name IS name
    children [
       ...
    ]
  ...
```

So when loading it in R2023b, we skip the scale field of the Solid and so the toilets are huge.

I don't think we can really do something about it.

closed 2023-06-16 by BenjaminDeleze