cyberbotics/webots #6256: [ 2023a & 2023b - Linux ] : Rendering view has black borders

SuperUserNameMan, 2023-06-27:
**Describe the Bug**
The rendering view has top and right black borders. (see screenshot)

**Steps to Reproduce**
Install 2023b or 2023a `.deb` or `.tar.bz2`

Open any sample demo/example.

**Expected behavior**
No useless black borders.

**Screenshots**
[image]

**System**
 - Operating System: LinuxMint 21.1 Vera (based upon Ubuntu Jammy) + MATE 1.26.0
 - Graphics Card: integrated Radeon GPU that comes with the Amd Ryzen 5 5500u APU.
``` glxinfo :
    Vendor: AMD (0x1002)
    Device: RENOIR (renoir, LLVM 15.0.3, DRM 3.48, 5.15.0-75-generic) (0x164c)
    Version: 22.3.0
```

**Additional context**

```
Graphics:
  Device-1: AMD Lucienne vendor: Hewlett-Packard driver: amdgpu v: 5.18.13 pcie: speed: 8 GT/s
    lanes: 16 ports: active: eDP-1 empty: DP-1,HDMI-A-1 bus-ID: 04:00.0 chip-ID: 1002:164c
  Device-2: Chicony HP Wide Vision HD Camera type: USB driver: uvcvideo bus-ID: 1-3:2
    chip-ID: 04f2:b735
  Display: x11 server: X.Org v: 1.21.1.4 compositor: marco v: 1.26.0 driver: X:
    loaded: amdgpu,ati unloaded: fbdev,modesetting,vesa gpu: amdgpu display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96
  Monitor-1: eDP res: 1920x1080 dpi: 142 diag: 394mm (15.5")
  OpenGL: renderer: RENOIR (renoir LLVM 15.0.3 DRM 3.48 5.15.0-75-generic)
    v: 4.6 Mesa 22.3.0-devel direct render: Yes
```

SuperUserNameMan, 2023-06-27:
PS : last known version to not exhibit this black border is 2022a

omichel, 2023-06-27:
It seems to be the same issue as #6055.
Maybe reverting [this patch](https://github.com/cyberbotics/webots/pull/5043/files) may help. Can you confirm?

SuperUserNameMan, 2023-06-27:
Thanks for pointing me.

Alas, reverting was not enough. 

I had to force `QT_ENABLE_HIGHDPI_SCALING` to 0.

omichel, 2023-06-27:
That probably means we need to handle your specific case for the setting `QT_ENABLE_HIGHDPI_SCALING` to 0.
Can you tell us with is the value of `XDG_CURRENT_DESKTOP` and `DPI` in your case?
Ideally, we should improve this script to take into account your specific case, as we cannot fully remove this code that would likely break other systems.

SuperUserNameMan, 2023-06-27:
According to what I read, the MATE desktop handles HiDPI displays in a specific manner that might differs from other linux desktops.

I plugged my laptop to my 4k TV and tested with HiDPI enabled and disabled in MATE display settings.

According to my tests the patch should be :

```bash
# Fix for i3 window manager not working with Qt6
if [ "$XDG_CURRENT_DESKTOP" == "i3" ]; then
  DPI=`xrdb -query -all | grep Xft.dpi | awk '{print $2}'`
  if [[ "$DPI" -gt 96 ]]; then
    export QT_ENABLE_HIGHDPI_SCALING=0
    export QT_SCALE_FACTOR=2
    export QT_FONT_DPI=80
  fi
# Fix for MATE desktop
elif [ "$XDG_CURRENT_DESKTOP" == "MATE" ]; then
  export QT_ENABLE_HIGHDPI_SCALING=0
else
  export QT_ENABLE_HIGHDPI_SCALING=1
fi
```

Note : if a user toggles the MATE's HiDPI setting during a session, it is required to disconnect from the session and login again (else, Qt6 or webot will not take the new setting into account).

closed 2023-06-27 by omichel