cyberbotics/webots #6257: SUMO extern controller and net-convert

BenjaminDeleze, 2023-06-27:
In the snap, we should now use the `sumo_supervisor.py` as an extern controller.

It works fine if the `useNetconvert` option of the `SumoInterface` is set to `TRUE`, like in `sumo_interface_example.wbt`.

However it seems like when launching the `sumo_supervisor.py` as an extern controller, it does not get the options set by the `SumoInterface` proto such as `useNetConvert` or `gui`.

So if you launch `city_traffic.wbt` with the `SumoInterface` as an extern controller, it will not work.

You have to manually specifiy: 
```
./webots-controller projects/default/controllers/sumo_supervisor/sumo_supervisor.py --no-netconvert
```

So, I guess that it is normal that the extern controller do not get the controller args set by Webots.
However, from a user perspective, it seems strange that the parameters are set in the proto, in the world but that is does nothing.

- Is it the normal behavior?
- Can we do something about it or not?
- If not, we shall document it, at least for SUMO because it is not clear for now.

stefaniapedrazzi, 2023-06-27:
I don't think there is any other solution than documenting this both in the PROTO as JS warning and in the documentation.

omichel, 2023-06-27:
I believe it would also be possible that the sumo_supervisor as a supervisor could get this information (controller args) by querying the scene tree and applying them.

Ideally, on the long term, we should modify the extern controller launcher so that it receives the controller args from Webots and applies them to the controller in addition to the args passed as a command line to the launcher.

closed 2023-06-27 by BenjaminDeleze