cyberbotics/webots #6266: Can't select Transform node in 3D View

ShuffleWire, 2023-06-28:
**Describe the Bug**
ditto
Especially bad because I can't move it, to do so I need to select it in the tree.

**Steps to Reproduce**
Open 
```
#VRML_SIM R2023a utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "webots://projects/objects/floors/protos/RectangleArena.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.36630783562978425 0.2247870123314165 0.902933756508936 2.12964342923002
  position 0.9655655585952454 -1.6769303338209047 1.9200352561851333
}
TexturedBackground {
}
TexturedBackgroundLight {
}
RectangleArena {
}
Solid {
  translation -0.24 0 0
  children [
    Shape {
      appearance PBRAppearance {
      }
      geometry Sphere {
        radius 0.1
      }
    }
  ]
}
Transform {
  children [
    Shape {
      appearance PBRAppearance {
      }
      geometry Box {
        size 0.1 0.1 0.1
      }
    }
  ]
}
```
Click on the sphere, the Solid is selected
Click on the cube, the Transform is not selected
Idem when using a Pose instead of the Transform

**Expected behavior**
Object should get selected.

**System**
 - Operating System: Webots master@5e8af90ac90158ef640a47ab932f8d7727da7594

**Partial solution**
Is the bug hiding near `void WbView3D::selectNode(const QMouseEvent *event)` or `void WbSelection::selectPoseFromView3D(WbAbstractPose *p, bool handlesDisabled)` ?

stefaniapedrazzi, 2023-06-28:
This is not a bug but a design choice.
Only Solid nodes can be selected from the 3D view.
If you want to move a Transform node using the handles in the 3D view, you have to select it from the scene tree.

The reason of this design choice is to make it easier to select the part of the robots that are commonly defined using Solid nodes.

ShuffleWire, 2023-06-28:
I see !

ShuffleWire, 2023-06-28:
BTW : reading the code I've seen 
`mSelectedAbstractPose = mSelectedNode ? dynamic_cast<WbAbstractPose *>(mSelectedNode) : NULL;`
could it be replaced by 
`mSelectedAbstractPose = dynamic_cast<WbAbstractPose *>(mSelectedNode);` ?
it seems safe according to https://stackoverflow.com/questions/5155820/portably-safe-to-pass-null-zero-to-dynamic-cast
but I'm not a C++ expert !

closed 2023-09-07 by ShuffleWire