cyberbotics/webots #6276: Black border in rendering view when using fractional display scaling on Linux

OPerepadia, 2023-06-29:
**Describe the Bug**

The bug happens on R2023a and R2023b.
When I set the global display scaling in system settings to 125% or 150%, I notice black borders appearing on the right and top of the rendering view. However, when I use 100% or 200% scaling, everything looks fine.

I  checked the similar issues (#6256 and #5024), but unfortunately none of the suggested solutions worked for me.

However, I found a combination that resolved the issue in my case. Apart from setting `QT_ENABLE_HIGHDPI_SCALING` to 0, I also had to clear the `QT_SCREEN_SCALE_FACTORS` variable (which was not mentioned in the related issues).

```bash
export QT_ENABLE_HIGHDPI_SCALING=0
export QT_SCREEN_SCALE_FACTORS=
```

**Screenshots**

Before
[image]
After
[image]

**System**

Arch Linux, KDE Plasma 5.27.6

omichel, 2023-06-30:
Thank you for this bug report!
Can you please propose a patch to the Linux launcher in a pull request similar to #6261?
Your patch must be very specific to your system (KDE Plasma), so that it won't break other users' configurations.
I just sent you an invite to become a member of the Webots committers group, so that you don't have to fork this repository to make your contribution and you will be able to merge your PR once approved.

OPerepadia, 2023-06-30:
Thank you, I will try to propose a patch.
There is one downside of this fix though. It requires high DPI scaling to be disabled, which makes the text in scene tree smaller, and changing the `QT_FONT_DPI` doesn't seem to affect it.
It still looks perfectly readable on my 24 inch monitor, but might be an issue with small laptop screens with high dpi.
However, the fonts in other elements, like the top Menu bar, context menus, Terminal, and Text editor are scaled properly.