cyberbotics/webots #6277: Overlays devices hiden or not

BenjaminDeleze, 2023-06-29:
**Describe the Bug**
Under the `Overlays` tab, the `Camera devices`, `RangeFinder devices` and `Display devices` sub menu are not always displayed.

It seems that, now, we need to select a robot and then go to the tab to see the actual sub menus.

Whereas before (in R2021b for example) the behavior was a bit more subtle:
- If there were several robots: you had to click on one to see the corresponding sub menus appeared.
- If there was only one: the robot was considered as selected and its sub menus appeared.

I think it is a regression but maybe it was done on purpose but if so, why?

Moreover, when going in the `Overlays` tab, it is not clear for me that the Devices are the one of the selected robot. As it is a general tab of Webots I would expect to see all the cameras and other devices of the world. However I think that it may be overwhelming, but we should at least change the name of the sub menus to make it clearer.

omichel, 2023-06-30:
I believe this is a regression and we should fix it.

stefaniapedrazzi, 2023-07-03:
This functionality was explicitly removed in #4350 (see description for the motivation).

I will check if it is feasible to show the overlays options for all the robots in the world.