cyberbotics/webots #6278: Problem in using assets offline

nchygy, 2023-06-30:
**Describe the Bug**
A clear and concise description of what the bug is.
Hi, as it's described in the USER GUIDE document, I am tring to download all the assets to use it offline.
My procedure as follow:
1. I download webots assets file named "assets-R2023b", decompress it and change the folder name to "assets"
2. copy that "assets"  folder with all the file included to the path "C:\Users\gy\AppData\Local\Cyberbotics\Webots\cache\"
3. restart webots and choose a sample in guided tour
4. click the 'User guide' button through webots interface menu
Problem:
1. webots can not properly use the assets and gives me an error like "ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/road/protos/Crossroad.proto': error: Template engine error: Cannot change directory to: 'C:/Users/gy/AppData/Local/Temp/webots-1234/'"
2. When I click the 'User guide' button through webots interface menu, webots is still open the browser and link to github.
Environment
 windows 11, webots 2023b, WIFI connected

**Steps to Reproduce**
1. Go to 'help menue.'
5. Click on '.Webots guided tour'
6. Scroll down to choose  'pr2.wbt'
7. See error :"Template engine error: Cannot change directory to: 'C:/Users/gy/AppData/Local/Temp/webots-1234/'"

**Expected behavior**
1. I should see the simulation with all the texture shown with out any connection to github
2. I should be able to read the 'User guide'  or "Reference manual" offline
I attach my assets folder as follow
Also I attach the screen of the error as the second figure
**Screenshots**
[image]

[image]

**System**
 - Operating System: [e.g., Windows 11]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 1080 11 GB]

**Additional context**
Add any other context about the problem here.

omichel, 2023-06-30:
You are describing two problems here:

1. The user guide (or reference manual) do not display properly from https://cyberbotics.com/doc/guide/index.
2. Webots fails to instantiate some PROTO nodes because it cannot access directory `C:\Users\gy\AppData\Local\Temp\webots-1234\`.

Here are my answers regarding the two problems:

1. The Webots documentation is hosted on GitHub and the web site of Cyberbotics is making requests to GitHub (possibly via jsdelivr) to display the contents of the Webots documentation. Therefore if you have some firewall blocking requests to GitHub, the documentation will not display properly. The solution is to use a different network location or a VPN to work around your firewall.
2. This problems doesn't seem to be related with the assets or with the cache. It is simply that for some reason Webots [cannot access](https://github.com/cyberbotics/webots/blob/96c96924c6ae6cc309f85b93ffc088c62affbc47/src/webots/core/WbTemplateEngine.cpp#L177-L181) the `C:\Users\gy\AppData\Local\Temp\webots-1234\`. Normally, when you start Webots this folder should be automatically created by Webots and when you quit Webots, it should be deleted.
    2.1 Can you check that this folder doesn't exist when Webots not running ?
    2.2 If it exists, please delete it.
    2.3 Start Webots and check that this folder exists. If it doesn't exists, this means that for some reason, Webots unable to create it and it should display an error message about it. You may want to check the permissions of the upper folder, e.g., `C:\Users\gy\AppData\Local\Temp\` and try to create manually a sub-folder in there.

Please try the above and report here about your results.

*Note*: I tested the installation of the assets as explained in the user guide and running Webots without any Internet connection and it worked well on my Windows 11 machine.

nchygy, 2023-07-04:
Hi
I followed your suggestion. Here are some replies.
1. Yes I know the documentation is block by the fire wall. However I still suggest a pdf version sould be availble for downloading because VPN is not always available and internet access may also not always be available.
2. I checked the file ‘C:\Users\gy\AppData\Local\Temp\webots-1234\’. It has been successfully created by webots as you described and deleted automatically once webots has been deleted
3. I also tried to creat an folder under  ‘C:\Users\gy\AppData\Local\Temp'. It succeed.
4. Inspired by your answer, I tried to running webots as a administrator, then the problem has been fixed.

closed 2023-07-04 by omichel