cyberbotics/webots #6285: Unexpected exit in "sumo_supervisor"

ZhuangshengLi, 2023-07-02:
version: R2021b
sumo fails to generate vehicles in `"highway.wbt"` and `"highway_overtake.wbt"`

When the `"highway.wbt"` had been loaded, the error is as follows:

```
C:\Program Files (x86)\Eclipse\Sumo\tools\sumolib\net\__init__.py:290: UserWarning: Module 'rtree' not available. Using brute-force fallback.
  warnings.warn("Module 'rtree' not available. Using brute-force fallback.")
Traceback (most recent call last):
  File "C:\Program Files\Webots1\projects\default\controllers\sumo_supervisor\sumo_supervisor.py", line 215, in <module>
    controller.run(options.port, options.disableTrafficLights, directory, 

  File "C:\Program Files\Webots1\projects\default\controllers\sumo_supervisor\SumoSupervisor.py", line 550, in run
    self.get_vehicles_position(id, self.traci.vehicle.getSubscriptionResults(id),

  File "C:\Program Files\Webots1\projects\default\controllers\sumo_supervisor\SumoSupervisor.py", line 190, in get_vehicles_position
    sumoPos = subscriptionResult[self.traci.constants.VAR_POSITION]

KeyError: 66
WARNING: 'sumo_supervisor' controller exited with status: 1.
```

omichel, 2023-07-03:
Can you please try it with Webots R2023b which was released last week?

*Note*: you will have to [install the latest version of SUMO](https://cyberbotics.com/doc/automobile/sumo-interface#windows) as it is not any more included in Webots.

Shane-l-z, 2023-07-03:
We had to use R2021b for a working version. It is quite difficult to use the latest version.
Here is the sumo version: Eclipse SUMO sumo Version 1.17.0

We found that it worked well under the version Webots R2022a and SUMO 1.17.0

omichel, 2023-07-03:
All right. I am closing the issue. Feel free to re-open if needed.

closed 2023-07-03 by omichel