cyberbotics/webots #6286: Template importing error when adding new objects On Ubuntu

xnjust, 2023-07-03:
**Describe the Bug**
Template importing error when adding new objects On Ubuntu (After downloading Webots, open the new world for the first time and it will report this error)
Same as the issue [6060](https://github.com/cyberbotics/webots/issues/6060)
**Steps to Reproduce**
1. Click on 'adding new object or import a new object'
2. Scroll down to any new objects
3. Click 'import'
4. See error and the world's screen was black with nothing(Infact, click or any action like Drag, Enlarge etc in world window, i can see the part of model outline of the robot)

**Expected behavior**
A clear scene with robot and background.

**Screenshots**
[image]

**System**
 - Operating System: [Linux Ubuntu 20.04]
 - Graphics Card: [Cirrus Logic GD 5446]
 - By the way, OS is based on Shadowless Cloud Desktop which belongs to Alibaba Cloud

**Additional context**
Webots has been installed by ".deb". And i set the proxy for webots, so there is no problem accessing the Github content theoretically. Beacuse the user is in China, so if you can't contact me in time, please contact me by email xnjust@outlook.com or 378099789@qq.com

stefaniapedrazzi, 2023-07-03:
What is the user name on the system?
Could you please try to install Webots on an account where the user name contains just letters [A-Za-z] or numbers but no special characters and report if the error persists?

xnjust, 2023-07-03:
> 

hi, my username as follow(just contains ‘@‘ ‘/‘ and ‘_’):
[image]

I will try again in ubuntu 20.04 based on VMware.
I guess the reason of bug is ‘/‘?

stefaniapedrazzi, 2023-07-03:
Yes, the reason of the bug is very likely a special character in the username or path that is not escaped correctly and interferes with the PROTO loading.

stefaniapedrazzi, 2023-07-03:
The issue should now be fixed in #6288.
You can download a test version of Webots from the artifacts section at the bottom of the following page:
https://github.com/cyberbotics/webots/actions/runs/5442415963?pr=6288

Could you please report if this version is working correctly for you?

xnjust, 2023-07-03:
> The issue should now be fixed in #6288. You can download a test version of Webots from the artifacts section at the bottom of the following page: https://github.com/cyberbotics/webots/actions/runs/5442415963?pr=6288
> 
> Could you please report if this version is working correctly for you?

I will

> The issue should now be fixed in #6288. You can download a test version of Webots from the artifacts section at the bottom of the following page: https://github.com/cyberbotics/webots/actions/runs/5442415963?pr=6288
> 
> Could you please report if this version is working correctly for you?

I will start this work in about four hours.
Before test this version, how can i remove old version, Just “sudo apt uninstall webots”?(I install webots by using “sudo apt install./webots.deb”

stefaniapedrazzi, 2023-07-03:
Yes, to remove the old version installed from the Debian package you simply have to run
```sh
sudo apt remove webots
```

xnjust, 2023-07-03:
> Some time ago, we implemented some simple scenarios using [blockly](https://developers.google.com/blockly). If I remember correctly the idea behind it was to use the graphical programming language to code, but transform it into Python code in order to use it as Webots controller. If you run Webots using the [web interface](https://www.cyberbotics.com/doc/guide/web-interface), then you can also integrate the Webots 3D scene and the programming window in the same page.

Yes! My idea is to convert graphical code into python. But if realize this idea by using “robot window”, user will get a single web page for each robot.  How to develop a editor for graphical programming to replace the text editor ? I mean Embedded interface instead of single web page.

PR #6288 (2023-07-04): Fix PROTO load when a context path contains backslash
Fixes #6286:
backslash in paths passed to the template PROTO JS script are not correctly escaped producing invalid JS scrips that cannot load the PROTO models.
closed 2023-07-04 by omichel