cyberbotics/webots #6337: Discrepancy in UR5e Robot Motion Planning

niklasconbotics, 2023-07-21:
Hello,

I'm currently using the PyMP library for path planning with the UR5e robot. I've been basing this on the official ur5e.urdf file to compute the motion. However, I've noticed a discrepancy: despite confirming that the joint angles match those from the final state of the motion planning path, when I attach the GPS node, it shows an offset of around 2-4 cm.

Interestingly, my forward kinematics check with the urdf file indicated that the end effector is indeed at the exact position. This leads me to speculate that the urdf file and the internal .proto file, upon which the UR5e robot is built, might not align perfectly.

Could anyone suggest a way to convert the ur5e.proto file into a urdf format? This might help resolve the discrepancy I'm facing.

Thank you.

mrudorfer, 2024-02-16:
Hi,  this sounds like it might be related to an issue I had #6504

haidar996, 2026-11-07:
I checked the Webots UR5e against the official Universal Robots kinematics (UR5e DH parameters: d1 = 0.1625, a2 = −0.425, a3 = −0.3922, d4 = 0.1333, d5 = 0.0997, d6 = 0.0996), in the same conditions as you: gravity on, the 6 motors driven to 10 random targets, the joint angles read from the position sensors, and a `GPS` in `toolSlot`.

The GPS position matches the official forward kinematics at the measured joint angles within **0.5 to 1.3 mm** (mean 0.9 mm), once the frames are aligned (Webots R2025a; the geometry of `UR5e.proto` has not changed since 2023). So the PROTO and the official URDF do agree; an offset of a few centimetres comes from somewhere else. Things worth checking:

1. **Base frame.** The frame of the Webots robot is the `base_link` frame of the official URDF. The UR controller and the DH table use the `base` frame, which is rotated by 180° about z (x and y have the opposite sign). Mixing them gives errors of tens of centimetres.
2. **Which point is compared.** `toolSlot` is at the flange (`tool0` / `flange` frames of the official URDF). A GPS that is translated in `toolSlot`, or placed on a gripper, is measured at another point than the one the planner targets.
3. **Robot pose in the world.** `GPS.getValues()` is in world coordinates; if the robot has a `translation` or `rotation`, the GPS position must be expressed in the robot frame before comparing it with the URDF.
4. **URDF version.** A URDF of the UR5 (non-e) or with other parameters differs from the UR5e by several centimetres.

To get a URDF of the Webots model itself, a controller can call `getUrdf()` (`wb_robot_get_urdf`). Note that, until #7091 is released, it must be called with all the joints at 0 (see #6504, and the workaround in #6887), otherwise the current joint positions are taken as the zero of the joints.