cyberbotics/webots #6340: Webots can not import vehicles which are generated by SUMO

Growing1000, 2023-07-25:
**When I opened my wbt document, the SUMO started to run, but Webots can not import vehicles. Here are error reports.**

`ERROR: In order to import the PROTO 'CitroenCZeroSimple', first it must be declared in the IMPORTABLE EXTERNPROTO list.
ERROR: In order to import the PROTO 'TeslaModel3Simple', first it must be declared in the IMPORTABLE EXTERNPROTO list.
ERROR: In order to import the PROTO 'TeslaModel3Simple', first it must be declared in the IMPORTABLE EXTERNPROTO list.
...`

 **However, those examples like sumo_interface_example.wbt and highway_overtake.wbt can add vehicles automatically on my computer.**

omichel, 2023-07-25:
You should simply add [these lines](https://github.com/cyberbotics/webots/blob/c3e94a811430bd4f70ec694efe5e37f50c7fdeee/projects/vehicles/worlds/sumo_interface_example.wbt#L12C1-L26) at the beginning of your `.wbt` world file.

closed 2023-07-25 by omichel

Growing1000, 2023-07-26:
Thanks for your reply. With your suggestion, the problem has been solved.