cyberbotics/webots #6345: addTorque and addForce are doing the same

Jay-Song, 2023-07-27:
to whom it may concern,

I made two simple boxes attached by a hinge joint and applied a torque to one box.

But, the box was not rotating and just moved in the direction of the torque I gave.

I tested addForce but it showed the same result.

I think addTorque and addForce are using the same code or there is some problem in distinguishing addForce and addTorque.

Thanks.

ShuffleWire, 2023-07-27:
He is right, I think this demonstrate is point (two box link with a hingejoint, on which we apply a force, should not move)

[test_torque_force.zip](https://github.com/cyberbotics/webots/files/12180381/test_torque_force.zip)

Jay-Song, 2023-07-27:
[box_test.zip](https://github.com/cyberbotics/webots/files/12183401/box_test.zip)

Here is what I tested.

CoolSpy3, 2025-08-20:
It looks like these two test worlds actually demonstrate different things. @Jay-Song's demonstrates that `Node.addTorque` (`wb_supervisor_node_add_torque`) does the same thing as `Node.addForce` (`wb_supervisor_node_add_force`) in Python (see #6879), whereas @ShuffleWire's demonstrates that `wb_motor_add_force` does the same thing as `wb_motor_add_torque`, which (I think) is intended behavior. I have a memory of that being part of [the Motor docs](https://cyberbotics.com/doc/reference/motor), but I can't seem to find it now. At the very least, it meshes with all of the other motor methods being shared between `Motor` and `LinearMotor` with just the units changed.

closed 2025-08-20 by CoolSpy3