cyberbotics/webots #6347: Motor ， some questions about getForceFeedback().

gh18406468101, 2023-07-30:
When I use getForceFeedback() in Motor, it consistently returns values around + - 0.2, regardless of the computed joint torque from my control algorithm. Why is this happening, and what does this API measurement value represent?
[image]
[image]
[image]
[image]

lkx8421, 2023-08-08:
I have the same problem too. And looking forward to the author's reply :)

ShuffleWire, 2023-08-08:
Have a look there : https://www.cyberbotics.com/doc/reference/motor , at the wb_motor_enable_force_feedback chapter : 
```
the wb_motor_get_force_feedback (resp. wb_motor_get_torque_feedback) function does not measure:
    The force specified with the wb_motor_set_force (resp. wb_motor_set_torque) function.
```
Anyway, please share a full minimal world if possible :)

ShuffleWire, 2023-09-06:
@lkx8421 The author doesn't respond, maybe could you instead provide a minimal world showing your issue ?