cyberbotics/webots #6361: I click on a node in the scene tree ,, webots will crash and not respond.

xishengping, 2023-08-21:
My problem is that when I click on a node in the scene tree, webots will freeze and not respond. I seted --mode=pause, and when it is turned on in safe mode, webots also crashed.

[image]

ShuffleWire, 2023-09-06:
Please try with Webots 2023b.

If possible share your world (you can zip all the folder and upload here), and describe the exact sequence of action that trigger the crash.
Also describe your configuration (OS, CPU, GPU...) and copy paste here the content of the Help/OpenGL Information... window.

If you can, start webots from a command line and share any message displayed there.

Could you explain what is "safe mode " ?

xishengping, 2023-10-24:
World file, demo program crash video I put in the attachment。

 configuration Information：
- Operating System: [Windows 10]
 - Graphics Card: [NVIDIA GeForce RTX 3060 12 GB.]
- webots version : [2023a]

Steps to Reproduce
1. Go to sence tree Panel
2. Click on [random](https://dashi.163.com/webmail/read/java_script:;) node
3. Click on add node or add [node attributes](https://dashi.163.com/webmail/read/java_script:;)
4. See the error

[

https://github.com/cyberbotics/webots/assets/49152163/1de659d6-62c0-4ef5-b1e7-131325b65786

](url)[
Uploading
[pick_and_place.zip](https://github.com/cyberbotics/webots/files/13110211/pick_and_place.zip)
 webots_crashed.mp4…
(url)