cyberbotics/webots #6363: Weird collisions with ElevationGrid if x and y dimensions diff is 1

Inngress, 2023-08-23:
**Describe the Bug**
While using an ElevationGrid as a boundingObject in a Shape, if its x and y dimensions have a diff of 1 (e.g. `xDimension 3` and `yDimension 2`), the collisions with any object don't match with what is rendered (and not with what is displayed by the boundingObject optional view neither). At some places the object goes through the floor as if there was nothing and at others it gets stuck on invisible terrain. 
[image]
[image]
Also repeating the same tests tend not to give same results.

I tested out on a large scale of points with a map that just had a quarter of it elevated and the rest at a null or negative height and it looked like the boundingObject was kind of mirrored on `y` but no way to be sure since it is rendered at it's intended place. Plus the mirrored bounding object is not well defined, it is not a perfect mirror, there are strange holes and unclear boundaries everywhere while trying to make a robot drive on it on example.

TLDR: real collision is often at the opposite of where it is supposed to be.

Don't know if this could be related to  #4169

**Steps to Reproduce**
1. Download [uneven_terrain_test.zip](https://github.com/cyberbotics/webots/files/12418797/uneven_terrain_test.zip) world and load it with Webots 
3. Move the box around
4. See error

**Expected behavior**
boundingObject working as intended for any set of dimensions

**System**
 - Operating System: Linux Ubuntu 22.04
 - Graphics Card: NVIDIA GeForce RTX 3070 Ti laptop 8 GB
 - Webots 2023b

ShuffleWire, 2023-09-06:
I've tried your world and didn't seems to be able to reproduced it.

Your images seems to show that the plane is tilted, when your world you provided is not, is that correct ?
If so, could you provide a precise world, with **all the object** placed exactly in the configuration that trigger the bug, **including the camera position**, so that we can both record a video showing (or not) the bug, and compare apple to apple.

Inngress, 2023-09-08:
Well, that's kinda awkward... In the world I sent I forgot to uncomment the elevation maps heights... So the floor is indeed flat. My bad.

[Here](https://github.com/cyberbotics/webots/files/12560755/uneven_terrain_test.zip) is a corrected version with exact camera and all. 

I also join gifs to illustrate what I have with the box at different $x$ values
$x=12$
[image]
$x=7$
[image]
$x=3$
[image]

Also I noticed that it happens also with more than 1 dimension diff. Actually maybe any dimension difference.

ShuffleWire, 2023-09-08:
might be linked to https://github.com/cyberbotics/webots/issues/4169

closed 2023-10-11 by omichel