cyberbotics/webots #6365: Occuppied or empty

abzzall, 2023-08-27:
How can I check if position (x, y, z) is occupied or empty in webots?

abzzall, 2023-09-07:
I have imported OSM map with the importer and it imported buildings as the SimpleBuilding object.
Also, if I place the little sphere in this SimpleBuilding object it does not detect it. Even if I try to set the lidar sensor it detects only from outside. From inside it does not detect SimpleBuilding object even if the rays of the lidar intersects the corners of the SimpleBuildings.
 Also, I have tried to place a big sphere and added a touch sensor, it detects the SimpleBuilding object only when the center of this sphere is outside.

ShuffleWire, 2023-09-08:
Could you please share a world with one sphere (the size you like), plus only one SimpleBuilding (not a full OSM map), that show the behavior you describe.

Also Touch sensor can be a bit tricky to use, depending on the Shape of the bounding object...

I don't understand why you speak about Lidar. Could you elaborate ?