cyberbotics/webots #6369: SupervisorController: The process crashed some time after starting successfully.

mostmosthandsome, 2023-09-06:
**Describe the Bug**
When I'm going through tutorial 8 and using the C++ code of the controller, I get a warning.

``
WARNING: SupervisorController: The process crashed some time after starting successfully.
WARNING: 'SupervisorController' controller crashed.
``

**Steps to Reproduce**
Follow the tutorial for creating a supervisor controller and use the same code in C++,.

**Screenshots**
[image]

**System**
 - Operating System: Linux Ubuntu 22.04
 - Graphics Card: NVIDIA GeForce MX450

ShuffleWire, 2023-09-06:
Could your share directly the world and the controller you are using ?

mostmosthandsome, 2023-09-06:
Supervisor.wbt

```
#VRML_SIM R2023b utf8

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/floors/protos/RectangleArena.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/robots/sphero/bb8/protos/BB-8.proto"
IMPORTABLE EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/robots/softbank/nao/protos/Nao.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.15070614678938904 -0.22783958704145219 0.9619650616819073 5.078479354073404
  position -2.311085369150199 9.17554091297603 4.181411698558042
}
TexturedBackground {
}
TexturedBackgroundLight {
}
RectangleArena {
  floorSize 10 10
}
BB-8 {
  translation 0.286683 1.54848 -0.0407249
  rotation -0.05304775215287365 -0.039267638585485064 0.9978196172412362 1.3081604511483345
  controller "<none>"
}
Robot {
  name "my_supervisor"
  controller "SupervisorController"
  supervisor TRUE
}
```

Here is the controller( the same as the tutorial)
```C++
#include <webots/Supervisor.hpp>

#define TIME_STEP 32

// All the webots classes are defined in the "webots" namespace
using namespace webots;

int main(int argc, char **argv) {

  Supervisor *robot = new Supervisor(); // create Supervisor instance

  // [CODE PLACEHOLDER 1]
  Node *bb8Node = robot->getFromDef("BB-8");
  Field *translationField = bb8Node->getField("translation");

  Node *rootNode = robot->getRoot();
  Field *childrenField = rootNode->getField("children");

  childrenField->importMFNodeFromString(-1, "DEF BALL Ball { translation 0 1 1 }");
  Node *ballNode = robot->getFromDef("BALL");
  Field *colorField = ballNode->getField("color");

  int i = 0;
  while (robot->step(TIME_STEP) != -1) {
    // [CODE PLACEHOLDER 2]
    if (i == 0) {
      const double newValue[3] = {2.5, 0, 0};
      translationField->setSFVec3f(newValue);
    }

    if (i == 10)
      bb8Node->remove();

    if (i == 20)
      childrenField->importMFNodeFromString(-1, "Nao { }");

    const double *position = ballNode->getPosition();
    std::cout << "Ball position: " << position[0] << " " << position[1] << " " << position[2] << std::endl;

    if (position[2] < 0.2) {
      const double redColor[3] = {1, 0, 0};
      colorField->setSFColor(redColor);
    }

    i++;
  }

  delete robot;

  return 0;
}

```

ShuffleWire, 2023-09-06:
Your call 
`Node *bb8Node = robot->getFromDef("BB-8");`
give you a NULL value, hence reference to that on the following line crash the code.

You can't access BB-8 that way, because it's not part of the Supervisor Robot you added.

I don't know how the tutorial is made, but I would better remove your supervisor, and add the controller to BB-8, and make him Supervisor if need be (it look like so, looking at your code)

mostmosthandsome, 2023-09-07:
I tried both methods, but it still crashed and the bb8Node is still a nullptr.
I think the difference between the function `getFromDef` and `wb_supervisor_node_get_from_def` do not matter because the function `getFromDef` is defined as follow in Supervisor.cpp
```C++
Node *Supervisor::getFromDef(const std::string &name) const {
  WbNodeRef nodeRef = wb_supervisor_node_get_from_def(name.c_str());
  return Node::findNode(nodeRef);
}
```
As to the structure of the world, I don't know exactly where the problem is. But in the [tutorial](https://cyberbotics.com/doc/guide/tutorial-8-the-supervisor?tab-language=c++) there is a image of the scene tree which is the same as mine
[image]

ShuffleWire, 2023-09-07:
[image]

You Forgot to add a DEF into it's definition

GerardHarkemaAvans, 2023-09-07:
Found it! Thanks

closed 2023-09-08 by mostmosthandsome