cyberbotics/webots #6370: Lidar data incorrect

ShuffleWire, 2023-09-06:
**Describe the Bug**
Even in a basic configuration I get data from a Lidar being significantly wrong.
I've setup a world where a Lidar look perpendicularly to a flat surface.
I can manually compute the distance I should get, using basic trigonometry.
I'm comparing to the Lidar data and get error of more than 10mm.

**Steps to Reproduce**
1. Open world [measure_lidar.zip](https://github.com/cyberbotics/webots/files/12534812/measure_lidar.zip)
2. Build and run controller
3. See output : right values are the error in mm
Plotting it in the included Spreadsheet give : 
[image]

**Expected behavior**
Error should be negligible.

**Screenshots**
The actual display in the GUI confirm that issue :  
[image]
The point cloud is wrong.

**System**
I will try more system as soon as I have access to them, especially using real GPU (I have a 1650 at home). I'm getting this exact behavior at least on those : 

- Debian 12 
```
System: Linux 6.1.0-11-amd64 x86_64
OpenGL vendor: Intel
OpenGL renderer: Mesa Intel(R) UHD Graphics 630 (CFL GT2)
OpenGL version: 4.6 (Core Profile) Mesa 22.3.6
Available GPU memory: N/A
```
AND (with software rendering)
```
System: Linux 6.1.0-11-amd64 x86_64
OpenGL vendor: Mesa/X.org
OpenGL renderer: llvmpipe (LLVM 15.0.6, 256 bits)
OpenGL version: 4.5 (Core Profile) Mesa 22.3.6
Available GPU memory: N/A
```

 - WSL : Ubuntu 20
 ```
 System: Linux 5.10.102.1-microsoft-standard-WSL2+ x86_64
OpenGL vendor: Microsoft Corporation
OpenGL renderer: D3D12 (NVIDIA Quadro T1000)
OpenGL version: 4.2 (Core Profile) Mesa 22.2.0-devel (git-e8fc5cc 2022-06-22 focal-oibaf-ppa)
Available GPU memory: N/A
```

TonyWelte, 2023-10-22:
This happens because the shader that computes the depth does not compute the depth for the range [-fov/2, fov/2] but for the range [-fov/2+fov/n, fov/2-fov/n] (n being the horizontal resolution). The values for ±fov/2 correspond to the very edge of the depth image and not the value of the edge pixels.

The fix for this is fairly simple https://github.com/TonyWelte/webots/tree/bug-lidar-data-skewed. However, there is a related issue in the other dimension (when multiple layers are used) that is not that simple to solve.
[image]

With multiple layers:
[image]

I think it is caused by two issues:
- the same issue with the wrong angle being used to generate the point cloud (the same kind of fix can be used)
- the resolution of the depth image not being a multiple of the number of layers (there is no pixel line that perfectly corresponds to each layer). With hFOV: 1.5708, vFOV: 0.6, hRes: 9, numOfLayers: 5 there is no issue because the depth image used is actually 9 by 5 (but with hRes: 11 the image is 11 by 6). 

I am not sure about what's the best solution to fix the second point :
- interpolate the pixel rows to get values corresponding to the actual layer angle
- compute the  depth for the correct angle directly in the shader

ShuffleWire, 2023-10-22:
I have no way to give you more insight on the bug, other that testing it (I will take some moment tomorrow at work to see if your patch with one layer and idea on multi layer could allow me get better result)

I will follow closely this topic, especially if any Webots member give additional insight ! 

Thank for your work anyway :)

omichel, 2023-10-23:
> I am not sure about what's the best solution to fix the second point :
> - interpolate the pixel rows to get values corresponding to the actual layer angle
> - compute the depth for the correct angle directly in the shader

I would go for the second option as it seems it could run faster and be more straightforward.

omichel, 2023-11-06:
Fixed in #6417.

closed 2023-11-06 by omichel