cyberbotics/webots #6380: It always crash when I use it

Liuliusui, 2023-09-14:
Hello,
I am a student in school. When my experimental class required me to use webots to conduct experiments, after I downloaded and installed it, I started to use it initially, but the problem was that it would crash inexplicably. It hasn’t gotten better even after some time.
This is my computer configuration, and I am using win11
[image]

ali1hammoud, 2023-10-10:
I had the same problem before, the solution was to reinstall another version of webots.

Liuliusui, 2023-10-28:
But the coursework should be done through version 2023b. :(

omichel, 2023-10-30:
Did you try to download a [nightly build of Webots R2023b-rev1](https://github.com/cyberbotics/webots/releases)? It should be largely compatible with R2023b.

ShuffleWire, 2026-04-05:
@Liuliusui I expect that your course to be over by now, do you still have the issue.
If you want to continue to use Webots, I would encourage you to use the lastest version.
If you are confortable with that, please close the issue.
I'm scared we won't be able to provide support for your specific issue on this old version...