cyberbotics/webots #6381: How do I use the gyroscope of the robot khr3hv ？

hbuxiaofei, 2023-09-15:
The robot khr3hv doesn't seem to have a gyroscope.

ShuffleWire, 2024-02-20:
in the proto of the khr3hv, i can read 
`
      DEF KHR3HV_Accelerometer Pose {
        translation -0.0833 0.0345 0.0
      }
      DEF KHR3HV_Gyro Pose {
        translation -0.0833 0.0345 -0.024
      }
`
Which indicate that their are no actual accelerometer nor gyroscope on the bot.
What about adding them ? @omichel 
Replacing the previous code by 
`
      DEF KHR3HV_Accelerometer Accelerometer {
        translation -0.0833 0.0345 0.0
      }
      DEF KHR3HV_Gyro Gyro {
        translation -0.0833 0.0345 -0.024
      }
`
should make it, at least for the OP usage
just checked but the khr2hv don't suffer the same problem, but maybe other could, idk

omichel, 2024-02-20:
Yes, feel free to propose a patch for this.

ShuffleWire, 2026-04-05:
fixed by https://github.com/cyberbotics/webots/pull/6959

closed 2026-04-05 by ShuffleWire