cyberbotics/webots #6387: macOS: No error generated when webots-controller cannot find executable

matl-hsk, 2023-09-22:
**Describe the Bug**
webots-controller prints "starting matlab" even if no matlab executable is found (at least on Mac).

**Steps to Reproduce**
1. Ensure that webots-controller does not find the installed matlab version (e.g. not on the global path) and don't specify it explicitly.
1. Start webots-controller with a matlab controller.

**Expected behavior**
An error should be printed that no matlab version was found.

**System**
 - macOS 13.4.1
 - Matlab 2023a

**Additional context**
See discussion in #6366

nilsjor, 2023-09-22:
Could I ask you to provide the full path to MATLAB on your machine? By default, it will look for it at:

* /Applications/MATLAB_R20XXx.app

As per the docs https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-language=matlab&tab-os=macos#running-a-matlab-extern-robot-controller

Is your installation located there?

matl-hsk, 2023-09-22:
That is correct. In the webots gui it works as intended and can find the installed Matlab version.

nilsjor, 2023-09-24:
Please tell me if I am on to something with #6388. I have no experience with macOS, and I have no machine to test it on.

matl-hsk, 2023-09-25:
unfortunately this does not help.

nilsjor, 2023-09-25:
It seems that you have the Application in the *correct* place -- just that the automatic invocation string is formed incorrectly. This is why no errors are given, because a valid installation is found.

1. Can you please provide the exact "matlab-path" arg that results in sucessfully running? 
2. Is the .app filetype necessary? What happens if you omit it?
3. If you manually provide an **incorrect** path, what happens then?

matl-hsk, 2023-09-25:
Thought it is probably faster if I just fix it myself. Thereby I realised that the selection process of the latest matlab version contained a bug, as well. Both are fixed with my PR.

omichel, 2023-10-16:
I assume we can now close this issue. Feel free to re-open if needed.

closed 2023-10-16 by omichel