cyberbotics/webots #6395: Wrong relative Position of objects in recognition by spherical cameras with pi/2 < fov < pi .

TheMangalex, 2023-09-28:
**Describe the Bug**
Hello everyone, I have noticed some problems with recognition objects in spherical cameras:

The position of an object in a recognition with a spherical camera becomes wrong if the center of the object/or a bigger part of it isn't  in the camera frustum anymore. This happens for fovs < pi. For fov > pi this seems to be fine. (Tested with fov of 2.14 and 3.0)
The wrong object positions during this projection force wrong bounding boxes and a wrong position in the webots_ros/RecognitionObjects Message.

Potential reason might be the mIsOmniDirectional bool in WbObjectDetection. 
The header has the annotation
`bool mIsOmniDirectional;  // is sensor omnidirectional (horizontal FOV < PI/2)`    (https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/utils/WbObjectDetection.hpp#L101 )
The cpp file implements it with
`mIsOmniDirectional(mHorizontalFieldOfView > M_PI),`   (https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/utils/WbObjectDetection.cpp#L40).

Changing this line to 
`mIsOmniDirectional(mHorizontalFieldOfView > M_PI / 2.0),`
lets it look correct for me, but I'm unsure if it is correct in all cases or just the two I tested.

I have some screenshots showing the problem and how it looks fixed attached below:

**Steps to Reproduce**
1. Use a spherical camera with fov between pi/2.0 and pi
2. Put an object close to the view frustum corner so that parts of it are outside it and parts are inside (as seen in the screenshots)
3. Use a recognition image/ros message
4. You can see the bounding box corners being wrong/the object position and other values being wrong

**Expected behavior**
The bounding box should match the recognition and the recognized object position should be correct. See screenshots below

**Screenshots**
Fov 2.14 before
[image]

Fov 3.0 before
[image]

Fov 2.14 after adjustments
[image]

Fov 3.0 after adjustments
[image]

**System**
 - Operating System: [Ubuntu 20.04 (Ros Noetic for Ros Messages)]

omichel, 2023-09-29:
Your patch seems good to me. Please open new Pull Request to propose your fix for this problem.

TheMangalex, 2023-09-29:
I opened PR #6396 for you.

closed 2023-10-02 by TheMangalex