cyberbotics/webots #6398: Panda gripper drops objects

Awais-Mughal, 2023-09-30:
**Describe the Bug**
I am using panda gripper on robotic manipulator and trying to pick a solid block, but the gripper drops the block. Would really appreciate any guidance to resolve this issue.

**Screenshot**

i have uploaded the recording of problem here:

https://www.youtube.com/watch?v=QQ8hE_REHgM

**System**
 - Operating System: [e.g., Windows 11]

**Additional context**
Add any other context about the problem here.

ShuffleWire, 2024-03-19:
Look like your video is not available anymore

Could you share your full world directly here ? 

Did you tune mass, coeeficient of friction, CFM, ERP ? that could help if well tuned, cause issue otherwise

ShuffleWire, 2026-04-05:
Don't have heard from you, I'm closing it, feel free to open it back if needed

closed 2026-04-05 by ShuffleWire