cyberbotics/webots #6407: webotsw.exe not running at all despite having all the requirements

PhobetorES, 2023-10-07:
**Describe the Bug**
After simply following the instructions and installation, the application won't run at all. not even crashing after opening up. when clicked nothing happens at all. tried reinstalling, disabling virus guard, updating drivers, running as administrator, setting the 3D settings in Nvidia control panel. Nothing worked out.

**System**
 - Operating System: Windows 11
 - Graphics Card: NVIDIA GeForce GTX1650 4 GB
 - Processor: intel core i5
 - RAM: 12GB

omichel, 2023-10-10:
Did you try the [safe mode](https://cyberbotics.com/doc/guide/starting-webots#safe-mode)?

PhobetorES, 2023-10-11:
> Did you try the [safe mode](https://cyberbotics.com/doc/guide/starting-webots#safe-mode)?

Didn't but found a way to be able to run

closed 2023-10-11 by PhobetorES

omichel, 2023-10-12:
> Didn't but found a way to be able to run

Can you share what was the problem and its resolution? It could be useful to others with the same issue.