cyberbotics/webots #6411: 3D simulation interface display issue

AXIBA-123, 2023-10-10:
**Describe the Bug**
When opening a world file（R2023b）, the 3D simulation interface may not display fully, with black screens on the top and right.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**System**
 - Operating System: [Linux Ubuntu 20.04]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 3060 6 GB, intel core i7-12700H 16GB, ]

**Additional context**
Add any other context about the problem here.

ShuffleWire, 2024-02-20:
Take a look at this https://github.com/cyberbotics/webots/issues/5024

ShuffleWire, 2024-03-19:
and more recently to that https://github.com/cyberbotics/webots/issues/6498#issuecomment-1939297732